SmartEVSE/SmartEVSE-3 #1: Unable to read modbus on V3

k-janssens, 2022-03-11:


koen-serneels, 2022-03-12:
@mstegen  I was also trying to get modbus to work to no avail, glad I found this issue in time. I'm in need to control the evse to work together with other devices consuming power, the evse without this feature to control it via any form of external api is a step backward for me.  

This feature was in v2 right? Can we perhaps flash v2 on the v3 device in the meantime?

Also, any other option of more high level communication than raw modbus (like a TCP or MQTT bridge) would be a huge step forward. If that is easier to implement than fix the modbus support than that would also be OK !

Would it be possible to give an ETA on this please as I'm really blocked atm wit this. Thanks!

mstegen, 2022-03-12:
I have ported v2 software to v3 hardware, but the ESP32 is completely different from the PIC which was used on v2.
The modbus code on v2 was written from scratch, on v3 i'm using a library, mainly because low level access to the uarts on the ESP32 (while running RTOS) is difficult. 
Until a few weeks ago it was for example not (?) possible to get a callback when new data was received by the uart.
You had to check for new data continuously in a loop. This is how the eModbus library determines there is new data.
10 days ago they added support for a callback (https://github.com/espressif/arduino-esp32/pull/6364) 
Hopefully it will get added to a stable arduino core release soon.

But in the mean time, i'll see if i can run the modbus master and node/slave processes at the same time.
That would most likely fix this issue.

koen-serneels, 2022-03-12:
Ok that would be great!

Another idea maybe, if running both processes would not be possible, is to have some kind of minimalistic firmware just driving the cp pwm and offer a modbus master for comms (or if possible a higher level tcp/mqtt bridge, who needs modbus anyways). This firmware could then be used by people controlling loadbalancing and/or power management themselves instead of relying on the evse/sensorbox.

mstegen, 2022-03-20:
I tested to run both the master and node processes, but that did not work.

So for now the only way to get it to work with a external master controller, is to configure the controller(s) as a Node, and send the required commands.

In order to keep each node happy, and not timeout with a comm-error you will have to send the chargecurrent for each node in a broadcast message to all nodes (address 09):
```
09 10 00 20 00 08 10 00 A0 00 00 00 3C 00 00 00 00 00 00 00 00 00 00 99 24
Node 0 00 A0 = 160 = 16.0A
Node 1 00 00 = 0 = 0.0A
Node 2 00 3C = 60 = 6.0A 
etc.
```
Each time this message is received on each node, the timeout timer is reset to 10 seconds. 
The master will usually send this message every two seconds.

This is all what is required to prevent the communication error.

To have full control over the nodes, you will have to read each node's status registers, and see if it requests to charge.

for example for node 2:
```Read status registers 03 04 00 00 00 08 F0 2E
Received packet (21 bytes) 03 04 10 00 01 00 00 00 3c 00 01 00 00 00 01 00 01 00 20 4d 8c
00 01 = state B
00 00 = no errors
00 3c = charge current 6.0 A
00 01 = Smart mode
etc.
```
Here the state changes to STATE_COMM_C (00 06)
`Received packet (21 bytes) 03 04 10 00 06 00 00 00 3c 00 01 00 00 00 01 00 01 00 20 0a 8e`
So the ESVE request to charge.

You can respond to this request by changing the state of the node to State_C
`03 10 00 00 00 02 04 00 07 00 00 49 D6`
Here it will write 00 07 (STATE_COMM_C_OK) to register 0x0000, and reset the error register 0x0001

The node will respond to this by switching to STATE_C (Charging).

k-janssens, 2022-03-20:
As I am now able to add an API endpoint to read and control the data I need, this feel like the better way to move forward.
So for me need anymore to access the EVSE through the modbus interface, not sure what the others their opinion is on this.

closed 2022-04-15 by k-janssens