SmartEVSE/SmartEVSE-3 #3: Question: How to setup dev environment?

k-janssens, 2022-03-13:
Not an issue but a question on how I (and others) can help growing this great product.

I've installed the V3 last week and it is working great (except for the modbus issue :)).

The only issue I've experiencing is the conflict between my inverter who wants to load the home battery, and the EVSE who wants to load my car battery.
The result is that the EVSE will always go to the highest current possible as the home battery will jump in if there isn't enough solar power.

So I was thinking of creating a PR for adding a API on your new webserver that will accept home battery current information to adjust the calculation of the car current.

I've already configured VSC with the PlatformIO plugin, but it seems the project.cmake file is missing.

So my questions:
- Do you have some documentation available on how to setup a local dev env (or some pointers, doesn't matter what format)?
- How to build the project and load this new software on the EVSE?
- For the software update, can I already use the /update endpoint you've added?
- Is there a way to locally test the software before uploading it to the EVSE?

mstegen, 2022-03-19:
I've got it running on both a Linux and Windows machine.
The steps to setup the environment are:

- Install Visual Studio Code
- Goto View - Extensions . Search for PlatformIO IDE. Install PlatformIO
- Git clone the SmartEVSE-3 repository on your local harddrive. (git clone https://github.com/SmartEVSE/SmartEVSE-3)
- click on the small 'house' icon at the bottom (blue bar) of your window.
- click on 'Open Project' , browse to the cloned repository. Note that you have to navigate to the SmartEVSE-3/SmartESVE-3 folder. (there should be a platformio.ini file in this folder)
- change the upload and monitor ports in the platformio.ini file (usually /dev/ttyUSB0 on linux or COMx on windows)
- now you should be able to build/compile the project (click the checkmark icon at the bottom of the window)

PlatformIO should install everything automatically.

please note that you need to patch the eModbus code in order for it to work correctly.

You can upload the code by pressing the upload button. This will only update the first code partition (ota_0). 
You can also use the /update endpoint for uploading the firmware, but this will update ota_1, and set the boot partition to ota_1. If you now upload the code with the USB cable it will update ota_0, so your changes are written to flash, but not active.
This can be very frustrating when debugging, so it's best to only use the USB method, or /update, but not both..

There is no way to locally test the software.

To answer your other questions,

k-janssens, 2022-03-20:
Thanks for the information!

One last question about doing an actual update:
If I use the /update endpoint and my code screws up the EVSE due to a coding error. 
As this update is written to the ota_1 partition, activating ota_0 should fix it?
How can I re-activate ota_0 again if the firmware on ota_1 is faulty?

mstegen, 2022-03-20:
Good question.
The ESP will do some tests if the partition is bootable, if not it will boot the previous partition.
But if you for example upload a firmware without /update feature working, then the only way to revert things is to erase the 'otadata' partition in flash.

This should work: (replace the COM7 port with the SmartEVSE's com port)

python esptool.py --chip esp32 --port "COM7" --before default_reset --after hard_reset erase_region 0xe000 0x2000

k-janssens, 2022-03-20:
Good to know I can revert, maybe need it today :)

You have an idea why the modbus isn't working anymore?

mstegen, 2022-03-20:
Modbus is working, as the master can read/write to the sensorbox, modbus kWh meters, and connected node SmartEVSE's.
It's just that the library i used does not allow an extra modbus master on the bus.

As long as there is one modbus master, and all the other devices are unique nodes/slaves it should work.

I'll see if i can write up a minimal set of commands, to send to each node for it to not display a comm error.

k-janssens, 2022-03-20:
No it is not about me using modbus with the EVSE. And now there is the possibility of adding web api this is less of an issue for me.

But on the EVSE screen I get this screen after the update:
[image]

Also on the status page there is not L1/2/3 data anymore.

k-janssens, 2022-03-20:
And would it be possible to upload the firmware.bin you used? This should also allow me to go back to the stock fw

mstegen, 2022-03-20:
Sure, here you go:

[firmware.zip](https://github.com/SmartEVSE/SmartEVSE-3/files/8311475/firmware.zip)

k-janssens, 2022-03-20:
No ... I've seen the issue but did not make the link I had to change this in the emod lib, all is working again, thanks!

k-janssens, 2022-03-20:
So I was able to add 2 functional endpoints:
- GET /settings --> to read the most important settings
- POST /settings --> to change the mode + provide EVSE with home battery charge data to take this into account when calculating EVSE current in Solar mode

Swagger contract: 
https://github.com/k-janssens/SmartEVSE-3/blob/feature/home-battery/SmartEVSE-3/evse-api.yml

It is still a WiP, but I can create a PR in time if this could be of interest to you?
https://github.com/k-janssens/SmartEVSE-3/tree/feature/home-battery

mstegen, 2022-03-29:
I have just checked your fork, Wow.
Great work!

I like that you got rid of websockets, and now uses ajax instead, it might not be the most responsive, but it's a lot more stable.
The Rest API is absolutely something i like to add.  So yes a PR is definitely appreciated. 
I will go through your changes in the next few days. 

Thanks again!

dingo35, 2024-05-31:
Added to the manual...

closed 2024-05-31 by dingo35