SmartEVSE/SmartEVSE-3 #4: Upgrade from v2

martijndierckx, 2022-03-13:
Is the RCM compatible with the one I purchased for v2?

wbitholedotbe, 2022-03-13:
Yes... It's the same.

mstegen, 2022-03-14:
Yes you can plug it in P1 (the white socket) on the SmartEVSE v3. It's not possible anymore to manually connect it to the 12 pin pluggable connector

bostjanlaba, 2022-08-08:
Is the main controller v3 different than v2? Can I upgrade v2 to work with wifi or do I need to purchase v3?

bobosch, 2022-08-08:
Yes, v2 is a different controller without WiFi

pascalsaul-copaco, 2023-03-11:
Can you also upgrade a v2 controller with the v3 software?

Ferforfaen, 2023-03-12:
I have just upgraded my V2 to a V3 (bought the V3 from Stegen) But to my suprise, my old sensor box does not work with the V3 hardware. Is this a known issue? Not keen to buy a sensorbox V2 as my old one works fine for my needs.

pascalsaul, 2023-03-12:
Can you also upgrade a v2 controller with the v3 software?

zharijs, 2023-03-12:
> Can you also upgrade a v2 controller with the v3 software?

Short:  No.

Long:

v2 runs on Microchip PIC MCU, v3 uses Esperessif ESP32 - one is small 8 bit MCU, other is dual-core 32bit SoC. The code base is radically different. Parts that are the same, are "software" things like menu, display commands, modbus protocol, but all hardware access, times, ADC access, port and peripheral access are different. 

Also, v2 using PIC, uses Microchip MPLab-X to IDE, ESP32 variant uses Arduino framework on top of PlatformIO on top of VSCode. So the toolchain is radically different too. 

So v3 software is what you would call a "port" of v2 software - soft parts are the same, hardware access and low level functionality is completely different.

mstegen, 2023-03-13:
Thanks @zharijs for the thorough explanation.

To reply to the question of @Ferforfaen , the protocol used on the sensorbox changed to modbus. You can reprogram the pic chip in the sensorbox 1 to make it compatible with SmartEVSE v3 (or SmartEVSE v2 with new software)
https://github.com/SmartEVSE/Sensorbox-1

If you can not reprogram the software yourself, send it to me, and i'll reprogram it for you.

avelinopepino, 2023-04-03:
Hi, I have a v2.12 SmartEvse with a sensorbox v1.5 and they work perfectly. I recently purchased a smartevse v3 and try to put it as master of the v2.12. It didn't work and as far as I know, I have to upgrade the v2.12. Is there any guidelines to do it for dummies?
Thanks in advance!

avelinopepino, 2023-04-04:
The issue has been somveolvedd. Please try to correct the order of RX and TX pins that are informed on the website.

closed 2024-05-24 by mstegen