SmartEVSE/SmartEVSE-3 #5: Read/Write values over HTTP

martijndierckx, 2022-03-13:
Since v3 includes WIFI and a webserver.
Is it possible to read/write the same values as we did over Modbus for the v2?

mstegen, 2022-03-19:
Currently the webserver just shows a status page. It's not yet possible to read/write values directly using unique endpoints like /chargecurrent or /temperature
The only two endpoints that currently work are /update (to update the firmware or spiffs partition) or /erasesettings to erase all settings, including the ESP's wifi settings.
This is something which is on my to-do list. Still figuring out what the best approach is...

martijndierckx, 2022-03-20:
The HTTP API would be my primary reason for upgrading my v2 to a v3. So keep me posted.

> Still figuring out what the best approach is...
Are you looking for an HTTP API architecture? If so ... I would be happy to help you with this.
(I could setup a spec in Postman)

fluppie, 2022-03-22:
Hey @martijndierckx check out @k-janssens in https://github.com/SmartEVSE/SmartEVSE-3/issues/3

martijndierckx, 2022-05-18:
@mstegen any update on this? I've got SmartEVSE3 sitting on my desk waiting to replace my v2. But I really need the HTTP API in order to proceed.

bostjanlaba, 2022-08-08:
There is a web server on v3? Does it support node.js? Is there any documentation so that I can create my own web service to control the charger, get data from it etc.? I did node.js api servers in the past but in this case I'm not sure how the api server itself reads, writes data from/to the controller itself. Is there a middleware for this?

Thank

martijndierckx, 2022-08-08:
I've switched to the forked [Serkri](https://github.com/serkri/SmartEVSE-3) firmware.
HTTP api is basic, but present.

mstegen, 2024-05-24:
All the work that went into Serkri's fork and more added by Dingo35 is now finally available here. Will close this now.