SmartEVSE/SmartEVSE-3 #6: WiFi disconnects after a day

k-janssens, 2022-03-16:
As title suggests, the WiFi connection stops working after a day or so.
On the EVSE I can see the message "not connected to WiFI", after cycling the power it immediately is back up and running.

There is an AP only a few meters away, rebooting the AP does not work.
Other devices connects without issues with this AP.

mstegen, 2022-03-16:
Ok, thanks for the bug report.
 i'll see if i can improve the Wifi reconnect functionality.

fluppie, 2022-03-18:
Maybe have a look here? https://savjee.be/2020/02/esp32-keep-wifi-alive-with-freertos-task/

k-janssens, 2022-03-20:
Not sure why but at the moment I didn't have this problem any more for the past few days

deqw, 2022-04-09:
See this WiFi disconnects too. Disabling and again enabling WiFi will do. Prefer automatic recovery or not disconnecting at all (or both ;o). Thanks for the good work!

Fixh1, 2022-09-27:
I'll try the 
WiFi.setAutoReconnect(true);
WiFi.persistent(true);
And see if it gives some improvement.

mstegen, 2022-09-27:
I have tried the 
WiFi.setAutoReconnect(true);
WiFi.persistent(true);
but it did not work, the controller would not automatically reconnect anymore.

I have not experienced wifi issues myself, so difficult to test.

Fixh1, 2022-09-27:
Hi @deqw,

Is it the disconnect handling that causes the problems, or the websocket part?
The first part is fairly easy, the last part is quit a lot of work...

Fixh1, 2022-09-27:
Hi @mstegen ,

I have experience a lot of problems with the WiFi, so testpossibilities enough :-(
I have build the SmartEVSE into an existing housing made from aluminium, so the Wifi is sometimes very poor.
Is it possible to connect an external antenna to the ESP32-WROOM? I've seen a youtube video where they connected an SMA connector.

deqw, 2022-09-27:
@Fixh1 I think they got rid of the websocket code in that serki fork, so if that is indeed the case it is the handling.
Maybe @k-janssens can shine a light on that.

Fixh1, 2022-09-28:
Yesterday I did a fruitless attemp with this code:

WiFi.onEvent(onWifiEvent);

void onWifiEvent(WiFiEvent_t event) {
    switch (event) {
        case WiFiEvent_t::ARDUINO_EVENT_WIFI_STA_GOT_IP:
            _Serialprintf("Connected to AP: %s\nLocal IP: %s\n", WiFi.SSID().c_str(), WiFi.localIP().toString().c_str());
            break;
        case WiFiEvent_t::ARDUINO_EVENT_WIFI_STA_CONNECTED:
            _Serialprint("Connected or reconnected to WiFi\n");
            break;
        case WiFiEvent_t::ARDUINO_EVENT_WIFI_STA_DISCONNECTED:
            if (WIFImode == 1) {
                _Serialprint("WiFi Disconnected. Reconnecting...\n");
                //WiFi.setAutoReconnect(true);  //I know this is very counter-intuitive, you would expect this line in WiFiSetup but this is according to docs
                                                //look at: https://github.com/alanswx/ESPAsyncWiFiManager/issues/92
                                                //but somehow it doesnt work reliably, depending on how the disconnect happened...
                WiFi.reconnect();               //this works better!
            }
            break;
        default: break;
  }
}

But that did not help. The connection is lost again.
Tonight I'll have a look at the websocket.
I think I will just remove it, and put all the variables in at generation of the page.
See if that stabilizes the connection.
The RSSI was -90, so very poor I think.

mstegen, 2024-05-24:
I hope the connection issues were solved. As the current software version is based on Serkri/Dingo35's work, and i have not seen this issue mentioned anymore, i'll close this issue.

closed 2024-05-24 by mstegen