SmartEVSE/SmartEVSE-3 #9: Wifi AP - default password

erikplestenjak, 2022-03-27:
Hi. I haven't been able to find a more appropriate channel/forum for my problem so I'm asking here as an issue.

So in the process of installing the SmartEVSE V3 I managed to lose the WiFi password that comes in the box with the device. There is no way of changing it in the interface. I tried connecting to It via CLI to check if it's possible to reset the password that way. I got the drivers installed on my Win11 laptop, and the device shows up on port COM3 when connected via USB. However I haven't had any luck connecting to it via putty with the settings 115200 8N1. 

Is there any way of changing the password other than re-flashing the device. And also, what are the parameters to connect  to the device via usb-serial on a windows machine?

adyredro, 2022-03-27:
Hello!  
If I'm not mistaken, you have to hold down the left and right arrows at the same time and the password and wifi settings will be reset.  
Good luck!

mstegen, 2022-03-28:
The WiFi AP password is generated by each controller at random when first powering up. And will be output on the debug output of the controller at powerup.
The serial settings are 115200 8N1, currently the output is read only. There is no CLI like on the SmartEVSE v2.

```
SmartEVSE v3 powerup
Checking eFuse Vref settings: OK
Total SPIFFS bytes: 534881, Bytes used: 193772
APpassword: e3068b6a

settings saved
```
The AP password is also visible on the LCD when you choose the option "Setup WiFi" (wait 10 seconds for the portal to start, and the password to be shown).
You should be able to connect to this AP with this password. 
From within the AP portal, you can  search for WiFi access points and connect the controller to the internet.

It's not possible to reset the password by holding the left and right buttons on the controller.

erikplestenjak, 2022-04-11:
The password did not automatically show on the LCD when selecting "Setup WiFi" and waiting. All it showed was "Connect to the Aceess point...". My EVSE has a SW version number 3.0.

The debug output via serial however did the trick. I had issues using Putty to connect, but once up and running, it prints it out directly.

For anyone in the future having problems for connecting via Putty, [this](https://stackoverflow.com/questions/27532527/cant-connect-to-serial-port-using-putty) was my problem:

`Putty wont open a connection from the "Serial" tab in the left column. Choose "Session" instead and select the checkbox for Serial.`

closed 2022-04-11 by erikplestenjak