SmartEVSE/SmartEVSE-3 #12: Length extension for "CP" line

kletteroli, 2022-04-14:
First I would like to send a big "Thank you!!" to all the contributors in this project!
I did not find a forum place to ask this non-code related question...

I would prefer very much to install the SmartEVSE-3 about 20 meters away from the car.
What seems to limit the distance is the CP line.
What type cable shall I give a try for carrying the PWM signal over that distance? 
Twisted pair? Coaxial? Or just common 1.5mm2, etc?

wbitholedotbe, 2022-04-15:
That's not going to go well... In my experience 10m is the limit. You will have problems when extending the CP line over that limit.

fluppie, 2022-04-30:
I have a +- 32m long charging cable. I use +- 22m XVB 5G6 and 10x 0.8mm² communication cable and then a 10m 32A 3Phase charging cable. So far (1.5months) it works. Only I do not get the current I want, but I think this is because of the length of the cable I get a lot of power loss. If I set 17A charging current I get +- 16-16.1A real current (I use a SDM230 kWh meter which measures the current through the contactor). At 32A set I get around 30.2A.
So short answer, it works, but you'll loose some power. For me the 10x 0.8mm² cable works well

bobosch, 2022-04-30:
On 22m 6mm² when charging with 32A the power loss is about 0,5A. Consider that also the car does not charge at maximum power. What cross-section (mm²) has the charging cable?

fluppie, 2022-04-30:
The 22kW 10m fixed charging cable is also 5G6 and 3x0.5. Got it from Stegen's webshop.
Is the loss linear? Which would mean the loss for 32m is +- 0.75A?
But the total charging loss in my case (BMW iX3) is pretty steep, between 16 and 23% loss. I calculate the kWh consumed from driven distance and average consumption reported by the car. Then see how much the SDM230 register while charging. I always charge back to 100%. I only don't know how much of this % is coming from the long cables and how much from the AC-DC converter in the car.
Anyway a bit offtopic... Goal was to let @kletteroli know it will work with longer cables :)

kletteroli, 2022-04-30:
Thank you very much, I will give it a try.
Do you have a reference (part number, etc or further description) for the 10 X 0,8mm2 com cable?

fluppie, 2022-05-07:
https://gigatek.be/nl/kabel-signalisatiekabel-svv-cca-10-x-0-8-mm-b500--kobsvv10x08

It's a really common cable. I think alarm installers use this a lot 4x0.8 / 8x0.8 / 10x0.8
I chose 10x0.8 because I also want to the RFID reader at 22m of the SmartEVSE controller. Still need to hook that up, so can't tell you if it works. But it should, as it's using OneWire and that usually can bridge long distances.

kletteroli, 2022-05-18:
CP is now connected with 15m 1.5mm2 plus a 5m charging cable. Total distance 20m to the SmartEVSE. This is the second test day and PWM transmission seems to work correctly. On solar mode the evening ramp down and the morning ramp up worked nicely.

closed 2024-05-24 by mstegen

fluppie, 2024-05-24:
I'm adding here that it's strongly recommended to use a shielded cable. I did an install where there where 2 EVSE's and the one CP was interfering with the other. After switching the cables (one time 30m and one time almost 50m) all issues where gone. So for new installs I'm always using shielded CAT6 or CAT7 cable.

mstegen, 2024-05-24:
Thanks. Good to know that when using shielded cable, you can even use it up to ~50m. I would have never expected that.

dingo35, 2024-05-24:
Ha Michael, we wisselen elkaar sf qua drukte: ik ben de komende dagen een keuken bij mijn dochter aan het zetten  geen idee hoe lang het duurt maar ik kom er op terug!

Hans