SmartEVSE/SmartEVSE-3 #14: 2 contactors and how it works

wbitholedotbe, 2022-04-18:
How does it work with 2 contactors.
How to connect them and what type to use.
Are we supposed to connect 2 times 2p contactors?
It's not clear to me. Thanks.

mstegen, 2022-04-21:
You can use two 2 pole contactors, for example the ISKRA 232-20.

The coil of contactor 1 should be connected to N and C1. This contactor will switch the N and L1 to the EV.
The coil of contactor 2 should be connected to N and C2. This contactor will switch the L2 and L3 to the EV.

The idea is that it allows for switching between one and three phase charging. 
There is also an analog switch built into the SmartEVSE v3 that can isolate the CP signal, so the EV thinks it's disconnected from the charging station.

Software to allow switching between 1 and 3 phase has not been written yet. But it should always disconnect the CP first, before switching on contactor 2. As the EV might not expect this, and this can cause problems/short circuit with the chargers in the EV. (The chargers are most likely switched in parallel with one phase charging)

deqw, 2022-04-24:
When using 2 pole contactors, can both be connected to C1 to enable 3 phase charging (while awaiting further development)? Or put a wire over C1 and C2?

mstegen, 2022-04-25:
Currently both outputs (C1 and C2) are switched on/off simultaneously, so you can already wire each contactor to it's corresponding output.

deqw, 2022-04-25:
Ah thanks, see it in your code now. I'm using the serki version at the moment which only switches C2 when using normal mode.