SmartEVSE/SmartEVSE-3 #16: Menu item missing for MainsMeterMeasure ?

fluppie, 2022-04-30:
Hi @mstegen,

I was browsing the code and think I've found a bug. When telling SmartEVSE what the mainsmeter is measuring, you only have two options/values in the menu, "All" and "Home+EVSE", while I think the code works/expects to work with 3 options/values.

Here you see the 2 options in the menu:
https://github.com/SmartEVSE/SmartEVSE-3/blob/c7c0ca81f03f97c1ba22a9f8a0a97deeeb02fe74/SmartEVSE-3/src/evse.cpp#L1382-L1384

Here you see 3 options:
0: Mains (Home+EVSE+PV) / 1: Home+EVSE / 2: Home
https://github.com/SmartEVSE/SmartEVSE-3/blob/c7c0ca81f03f97c1ba22a9f8a0a97deeeb02fe74/SmartEVSE-3/src/evse.cpp#L122

I also want to understand it right. I have 3 SMD230 meters, one which measures more or less the same than the kWh meter from the power company, so everything. Another one measuring every kWh that goes into the EV. And last one that keeps track of the kWh produced by the solar PV system.

When choosing option All, it sets MainsMeterMeasure to 0. Which I think is right, but then it disables the PV Meter and PV Meter settings. or I misunderstand the whole thing, or some menu settings was forgotten when solar mode was developed?

mstegen, 2022-05-05:
Hi,

The mainsmeter options were developed by @bobosch with a specific use case in mind. If i remember correctly the Home+EVSE option measures everything except PV which is fed into the grid -before- this meter.

```Possible positions of electric meter:

 Home
   |
  (C)
   |
   +--[SUPPLY]--(F)--+--(E)--[CABLE]-- Car 1 [MIN]-[MAX]
   |                 +--(E)--[CABLE]-- Car 2 [MIN]-[MAX]
  (B)
   |
   +------------(D)------------------- PV
   |
  (A)
   |
[MAINS]

(X): electric meter
[XXX]: current limitation

Configuration

- Main electric meter (reduces charging current in smart mode)
  - Type [None* / Sensorbox / Finder / Phoenix Contact]
  - Measures [Mains (A) / Home with EVSE (B) / Home without EVSE (C)]
```
You are right there was the option 'Home without EVSE' in the original idea, but never implemented.

I think in your use case you can just use the one SDM230 that is in series with your official energy meter, and one that measures the EV power/energy. 
There is no need to have a seperate PV meter.

The code will take the Power and Voltage measurements of your SDM230 and use that to calculate the current.

bobosch, 2022-05-06:
Hi! Yes, "All" is correct when the EM is placed in the same position as the EM from the power company. Then it must be able to detect the current direction. Some EMs always reports positive current values (Eastron as well), this works only with the predefined EMs. Therefore multiple registers are considered:
https://github.com/SmartEVSE/SmartEVSE-2/blob/ef703d897320e47cfccbcc3e98c45c7c4e7411f4/SmartEVSE2.X/modbus.c#L583-L594
https://github.com/SmartEVSE/SmartEVSE-2/blob/ef703d897320e47cfccbcc3e98c45c7c4e7411f4/SmartEVSE2.X/modbus.c#L722-L728

dingo35, 2024-05-31:
I took the liberty to remove this code because it turned out to be of no use to anyone...

closed 2024-05-31 by dingo35