SmartEVSE/SmartEVSE-3 #17: Enable or facilitate support for EVCC #44

hyperbart, 2022-05-04:
It would be nice if SmartEVSE could be made compatible with https://github.com/evcc-io/evcc .

EVCC would act as a charge controller so you don't have to put clamps in and it does this by combining all metered devices (solar inverters, digital meters etc) and combine this with the SoC of a car to efficiently charge one or multiple cars.

dingo35, 2024-05-31:
Closing topic because of inactivity.

closed 2024-05-31 by dingo35

stomko11, 2024-11-13:
Is there any update in regards to support in EVCC?

dingo35, 2024-11-13:
No, there are still no plans for special support; we have APIs to connect to every environment you want to, but you have to make some effort yourself.

"evcc ist ein Energie-Management-System mit Fokus auf Elektromobilität. Die Software steuert deine [Wallbox oder Schaltsteckdose](https://evcc.io/#devices). Um intelligente Entscheidungen zu treffen, kommuniziert es auch mit deinem [Fahrzeug, Wechselrichter oder Hausspeicher](https://evcc.io/#devices). Die Software ist Open-Source und lebt von der Community." 

Thats exactly what SmartEVSE is doing 😁