SmartEVSE/SmartEVSE-3 #18: Documentation Getting started

systeembeheerder, 2022-05-08:
The [readme](https://github.com/SmartEVSE/SmartEVSE-3) is lacking some basic getting started info IMHO. 

After listing the features it start with _Connect the SmartEVSE controller to your WiFi network (using the menu of the SmartEVSE)_ but connecting to the network is not possible using the menu. The menu only enables the hotspot. The password for the hotspot is only to be found using an usb3 cable, putty and the knowledge to use baud rate 115200.

mstegen, 2022-05-09:
I'll update the readme and manual with the information to connect the SmartEVSE to WiFi. 
The password for the hotspot is also visible in the top line of the SmartEVSE LCD.

systeembeheerder, 2022-05-09:
> top line 

User error on my side.  Totally missed the top line because the device is mounted pretty low. Thanks for the quick support @mstegen !

closed 2022-06-27 by mstegen