SmartEVSE/SmartEVSE-3 #19: Too high initial current draw immediately after plug in of EV

kletteroli, 2022-05-18:
The SmartEvse with FW. 3.0.1 and in Solar mode with -0A Startup. Mains at 10A and Max at 18A. Single phase.
On plugging in the car I register a current of aprox 18A on my mains which is of short duration (1-3 s). The expected behaviour would be not to surpass the 10 A mains limit. What can be done to avoid this? (I don't want to lower Max) 
Would it make a difference if I used a switch on the SmartEVSE to start/stop charging?

(For us this current spike is problematic as we live off grid with a battery inverter that is then entering into overload mode which is especially the case when the battery is full and when the additional PV inverters are (still) in idle mode and not yet contributing.)

mstegen, 2022-07-05:
Hi, 
This is caused by the EVSE slowly increasing the charge current, and the EV usually waiting 10-40 seconds before actually drawing current. 
I think the easiest solution would be to add a delay of let's say 60 seconds before increasing the charge current.
This way the EV can start drawing current, at the lowest setting (MIN) and this high initial current draw will not occur.

bobosch, 2022-07-05:
With the git version you could enable automatic StartCurrent detection - this will charge the first  40secs at 6 A

elas00, 2022-08-08:
Where can find firnware with fixed initial current issue?

kletteroli, 2023-02-27:
> Where can find firnware with fixed initial current issue?

The fork "kletteroli" has a fix. It is working for my Leaf/E-NV200.

dingo35, 2024-05-31:
Is this still relevant in v3.6.0?

mstegen, 2024-05-31:
Yes. I think it's good to re-introduce this in our current version.
So that it starts charging at 6A for 40s. And then starts increasing the charge current.

mstegen, 2024-06-06:
Here is a test firmware which will start charging at 6A (or MinCurrent) for 40 seconds, and then starts increasing the charge current. 
[firmware.zip](https://github.com/user-attachments/files/15677972/firmware.zip)

mstegen, 2024-06-19:
This has been implemented in v3.6.2
Closing this now.

closed 2024-06-19 by mstegen