SmartEVSE/SmartEVSE-3 #21: How to USB-C cable?

dingo35, 2022-06-08:
First of all: big thanks for this great project, love the hardware, love the software.

I have customized the firmware, and now finally got overconfident and messed up the firmware. I always used the wifi /update method, so as I understand ota1 now has my bricked firwmare, and ota0 still is supposed to have the original firmware.

My first problem is the USB-C connection: I see a 6 pin header marked P2, is this the header the USB-ftdi cable has to be connected to? What are the specs of the header-pins, or is there a standardized USB-cable that I can buy that has the header already in place?

I understand I can clear the ota partition with the esptool: 
python esptool.py --chip esp32 --port "COM7" --before default_reset --after hard_reset erase_region 0xe000 0x2000

Will this clear the ota0 (usb-programmable) partition or the ota1 (wifi /update partition) ?

Big thanks for your support!

dingo35, 2022-06-08:
Thanks for your quick response!

For all others out there, the USB-C port is located at the top right side, next to the high voltage connections.

Now the esptool command worked, but I expected the SmartEVSE to reboot on the original firmware. It did not, the LCD screen keeps flashing (like it did with my bricked firmware) and repowering does not change anything.

The vscode upload is not working, it cannot the device:
[code]Reading symbols from /usr/local/src/SmartEVSE-3-main.orig/SmartEVSE-3/.pio/build/release/firmware.elf...
done.
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-prog
PlatformIO: Initializing remote target...
Open On-Chip Debugger  v0.10.0-esp32-20210721 (2021-07-21-13:33)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.
adapter speed: 5000 kHz

Info : tcl server disabled
Info : telnet server disabled
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6014, description '*', serial '*' at bus location '*'

openocd: ../src/jtag/core.c:343: jtag_checks: Assertion `jtag_trst == 0' failed.
Aborted (core dumped)
.pioinit:11: Error in sourced command file:[/code]

The vid and pid are completely wrong, lsusb says 
Bus 003 Device 004: ID 10c4:ea60 Silicon Labs CP210x UART Bridge  .

What would be the esptool command to flash the current firmware (when I compile the current git)? Same for the spiffs partition? 

Also: I assume the SmartEVSE should be powered (no car connected), it will also power on via the usb connector....

dingo35, 2022-06-08:
Ok just found out that vscode does NOT upload the code automatically by running the debugger, for anyone else out there: click the platformio.ide icon, a strange out-of-space icon on the left menu bar, and there is an upload option available there. That works.

Now just one remaining question: if I now upload through the /update wifi method, I still boot on the ota0 usb partition (as you warned people for). How can I make the SmartEVSE switch to booting from ota1 again, so I can resume the comfortable wifi updating?

mstegen, 2022-06-08:
I just checked my program script, and i program the ota partition with a file called 'boot_app0' you can find this file in the 
.platformio\packages\framework-arduinoespressif32\tools\partitions> folder.
(i have also attached it)

python esptool.py --chip esp32 --port /dev/ttyUSB0 --before default_reset --after hard_reset write -z --flash_mode dio 0xe000 boot_app0.bin

You can also do a full erase from within platformio, but then you'll lose the per device keys. (note that these are currently not used)

The controller can be powered through the USB connection. That's how i program them.

[boot_app0.zip](https://github.com/SmartEVSE/SmartEVSE-3/files/8864695/boot_app0.zip)

dingo35, 2022-06-09:
Big thanks for your excellent support @mstegen !!

Just for collective learning: 
I did not need to flash the boot_app0.zip , I don't think the ota0 / ota1 works as you described (being: ota0 = flash by usb, ota1= flash by webserver, and your warning not to mix it up).

I don't know if this is relevant documentation:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/ota.html

..but it seems to work exactly this way: when you flash by webserver, it always is flashed onto the partition that is not in use, because it cannot flash over the running partition. Then at reboot it tries to execute the new firmware, if that fails it rolls back to the old firmware. Flashing by USB I didn't test, but I would expect it to behave the same way.

As for debugging, I added a library that allows remote debugging over wifi at serkri's github (trying to add my stuff to his github to prevent too many forks): https://github.com/serkri/SmartEVSE-3/pull/23  . I only use the serial print facility but it can even be expanded with breakpoints and manipulation of global variables, looks great!

cvanlith, 2024-01-31:
I needed to install the CP210x_Universal_Windows_Driver to be able to upload the firmware from platformio.

mstegen, 2024-05-24:
Closing this one @dingo35 ;-)

closed 2024-05-24 by mstegen