SmartEVSE/SmartEVSE-3 #23: Solaredge meter issue

elas00, 2022-07-09:
Connected to solaredge 3 phase inverter, shows - 3277A, mains meter configured as solaredge.
[image]

Any solution?

bobosch, 2022-07-09:
The Solaredge inverter uses other modbus registers than the Solaredge meter.
https://www.solaredge.com/sites/default/files/sunspec-implementation-technical-note.pdf
Try custom electric meter with the settings
"BYTE ORD" HBF & HWF (Byte order of custom electric meter)
"DATATYPE" 3 (Data type of custom electric meter)
"FUNCTION" INT16 (Modbus Function of custom electric meter)
"CUR REGI" 40072 (Register for Current (A) of custom electric meter)
"CUR DIVI" 1 (Divisor for Current (A) of custom electric meter)

bobosch, 2022-07-09:
But it will only measure the PV generation? So it will only show the PV current, but SmartEVSE will not work as expected as it needs the current from Mains

elas00, 2022-07-09:
Solaredge inverter measure mains via inepro r380 connected via modbus and is working fine on solaredge app.

bobosch, 2022-07-10:
Does the value -3277A change or is it always the same? Nevertheless please try custom EM. You can hold the button to fast change the settings. And going backwards will be a little bit faster because highest value is 65530 so only 25500 to reduce :-( Or use serial console.

elas00, 2022-07-11:
Value was always the same, now work ok with mains meter configured as solaredge.
[image]

closed 2022-07-11 by elas00

devdems, 2023-09-12:
@elas00 can you tell how you connected the charger to SolarEdge to get the data from the InePro meter?