SmartEVSE/SmartEVSE-3 #24: OCPP support?

fluppie, 2022-07-23:
Seems like libraries are being developed to add OCPP support to for example OpenEVSE
https://github.com/matth-x/ArduinoOcpp
https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/master/src/ocpp.cpp

Could SmartEVSE v3 also be OCPP compatible in the future?

dries007, 2022-08-09:
I would also appreciate this capability.
I now own a SmartEVSE v2 and would upgrade to a v3 if this becomes an option. 
Would merge requests be welcome for this?

mstegen, 2022-08-09:
Yes merge requests are welcome, once it's clear it's possible to include this into the SmartEVSE v3 software.

dries007, 2022-08-09:
Is it possible to develop on a generic ESP32 WROOM board? I'd prefer that over having to buy a second SmartEVSE before I know it'll be something I can use :)

mstegen, 2022-08-09:
Yes that will work (just tried), although i had to change the upload speed to 921600 bps.
v3 does not have a CLI option like v2 had, so it's not possible to configure it without display + buttons.

But of course you can force the Wifi-portal by setting WIFImode to 2. That should start the portal. 
The portal password should be listed on the serial debug output.

You can also send me an e-mail and i'll lend you a light used SmartEVSE v3 to dev on. ;-)

dries007, 2022-08-09:
I think the only thing I can't DIY from the schematic is the LCD, but a borrowed unit would help a lot. I'll email the address on the website.

fluppie, 2022-08-09:
Nice :-), OCPP could open the way to have automatic invoicing of home charging by parties like Last Mile Solutions.

wbitholedotbe, 2022-08-29:
I don't think it's possible without an internal clock

dries007, 2022-08-29:
I've used the network time sync as a clock source so far.

A small status update from me:

I'm having more problems with the way OCPP 1.x makes assumptions about the relationship between components (controller, server). It basically assumes the evse is dumb and all settings and configuration must be done via OCPP, meaning I'd have to strip out significant parts of the current firmware or make them work together somehow.
OCPP 2.x seems much better in this regard, since it acknowledges the existence of local load balancing for example. The problem is that I can't find a working (embedded suiteable) library for OCPP, and even most server projects are still on 1.x.

For now I'm focused on getting to know the standards and the existing firmware a bit more before I push stuff to git.

wbitholedotbe, 2022-08-29:
You should focus on ocpp1.6 as v2 is far to complicated and not widely used yet. Not to mention not every backend supports v2

mstegen, 2022-09-13:
Not sure if it's of any use, but EVerest has implemented the OCPP-J 1.6 protocol in a module.

It's available here:
https://github.com/EVerest/everest-core/tree/main/modules/OCPP

Timple, 2025-02-21:
I guess this can be closed. I veried it to be working!

closed 2025-02-22 by dingo35