SmartEVSE/SmartEVSE-3 #25: Support of ISO 15118 to read car SOC

abstractionnl, 2022-07-26:
At some public chargers (particularly ones by Alfen), the car can communicate with the charging station to give details about it's SOC, this is shown on the charger display. Looking around the internet this is done with the ISO 15118 standard. Having the SOC would allow some new smart additions:

 - Stop charging when battery reaches a certain % (to prevent battery degradation)
 - Switch to solar mode when battery reaches a certain % (always have enough available for small rides and only trickle in more on solar when available)
 - Charge just in time by setting a goal percentage to be available at a certain date/time, and wait for solar in the meantime.

mstegen, 2022-07-27:
I just heard about this a few days ago. These chargers talk to the car behaving like a DC charger, over ISO 15118 you can indeed request the SOC and other parameters of the EV.
After receiving this information they then basically tell the EV, oh sorry i'm not capable of delivering DC, and start charging in normal AC mode.
If you stop the charge at regular intervals, and then start the sequence again, you can predict when the EV is fully charged.

The SmartEVSE v3 does not have a ISO 15118 modem. So its not possible to request this information.

abstractionnl, 2022-07-28:
It looks like it is using powerline communication, over the CP wire. An external modem would be needed, but I see none available over the shelf.

fluppie, 2022-07-29:
See this paper: https://ltu.diva-portal.org/smash/get/diva2:1546405/FULLTEXT01.pdf
Or with this Qualcomm chip: https://www.qualcomm.com/products/application/automotive/auto-connectivity/qca7006aq

A workaround for the current SmartEVSE is integration with https://evcc.io to get SoC via BMW Connected Drive, MyAudi, Volvo on Call etc.
https://docs.evcc.io/docs/devices/vehicles

martijndierckx, 2022-08-04:
SmartEVSE v4? :-)

Just noticed the same at a Tesla Supercharger, charging my non-tesla. Also wondering: “If Tesla can read that data, than SmartEVSE should also be able to do it … right?”

mstegen, 2022-08-04:
Ha!
I'll see if i can order some pre-made modules with a Qualcomm chip. I found some from 8devices and chargebyte/intech.

Fixh1, 2022-09-27:
I added an MQTT task to the SmartEVSE (and installed Mosquitto, Node Red etc) And via Node Red I can get all information from my Zoë, including the SOC. So I can forward that information to the SmartEVSE. Not using it yet, but the option is there.
I only have a problem with the disconnecting Wifi :-(.

fluppie, 2022-09-27:
Yeah sure, but that's not always working well. Like BMW's connected drive that from time to time doesn't work for 1-2 days. The advantage of the qualcomm chip is it can always read SoC. Ideal for cars in an underground parking lot without wifi.

martijndierckx, 2022-09-27:
I noticed the same with Porsche's service. Not very reliable, plus on many occasions with delays.
So a direct integration would be much much better.

dingo35, 2022-12-23:
> 8devices

Not sure what the status is, but I would be happy to help develop the firmware adaptions for this; I am one of the developers of the serkri distribution, but it would make no sense doing that work twice....

fluppie, 2023-01-24:
@mstegen are you (still) looking into this for v4?

mstegen, 2023-01-27:
Thanks for the reminder, just ordered some red/yellow beet-e modules to play with.

martijndierckx, 2023-02-21:
@mstegen any luck on testing the beet-e?

mstegen, 2023-02-22:
I received the modules (Red beet/ Yellow beet and Red beet carrier board), but had no time yet to look into it.

synegic, 2023-03-03:
I am also actively looking for something that could 1. provide me with a SoC on the car and 2. authenticate the car without having to use a RFID card/manual action. It seems ISO15118 is the way to go forward.

I also stumbled across the web already in my quest. One thing that was very enlightening was this; https://github.com/uhi22/pyPLC#example-flow. I also came across Python libraries supporting both SLAC (link negotiation) and the actual ISO15118 communications. It all looks pretty expansive, but very cool! I mean come on, talking over IP with your car, that's awesome!

I am not sure if implementation directly inside SmartEVSE is a good idea, though. It would most likely be possible, but full integration (including the Plug&Charge stuff) would require quite a big list of 'extras', like TLS, IPv6 support, UDP, TCP servers, etc. It sounds like a lot to handle for the ESP32 currently used.

With that in mind, I'd like to try to add ISO15118 through a separate component (e.g. Raspberry Pi) that could act as a charge-controller-controller, handling the actual powerline communication with the car and 'work together' with the SmartEVSE to control the contactor, as the basic communication over the CP/PE lines (figure out the EVSE 'states') is still required.

I hope the control signalling on the CP line (e.g. load balancing) and the PLC communication are not xor, and can be combined. Else above may not work out.. But we can at least try!

I have sent a mail to the people over at CODICO to find out if that carrier board they have listed on their site containing the RED BEET (https://www.codico.com/de/red-carrier-board-e-1-1-evse) would be plug&play and allow me to get a head start.

ArendJanKramer, 2023-03-09:
@arpiecodes I'm pretty interested in what CODICO has to say. Thanks for the link though, I hadn't spot the carrier boards before.
I ordered one, will update you all. Like you said, pretty awesome to talk over IP with the car.
I have also seen this one: https://www.codico.com/en/powerline-gp-evse-din-rail-modul-version-1-1
But holy smokes, that's not cheap for DIY.

My first plan is to setup a Raspberry PI with the board, and see if we can get some form of communication. Things I'm interested in are:
 - First of learning more about this topic, I'm a blank slate
 - Read car information, perhaps useful for authorization purposes
 - Read SoC and other information. 

I also looked at the carrier board with a white beet instead, which has an ST32 microcontroller. However, if I understood correctly the sample code needs to be licensed, and I couldn't find sample code online. 
Otherwise, I'd like to have experimented with that instead.

Once this all works, we could make a PCB with small microcontroller and a RED beet that acts as a modbus slave for SmartEVSE to supply information about car SoC. (or SmartEVSE v4)

One thing I'm interested in to learn is how we deal with CP signal, as it is currently controlled by SmartEVSE, while pyPLC mentions that the PWM needs to be set to 5% duty cycle for a while to setup digital communication. Well, we'll see once the board arrives and things need to be wired. I usually learn these things from doing it wrong initially.

dingo35, 2023-03-09:
Since the SmartEVSE already controls the CP line, and already has a full tcp server running (e.g. serkri distribution), IMHO it would make most sense to connect the modem through the existing SPI interface on the SmartEVSE board, and do all low-level communication on the ESP32; it might be necessary to offload the tls and complicated encryption/decryption routines that are involved with ISO15118 to a raspberry pi, but doing ipv6 is already possible since 2019 so that shouldnt be a problem...
Also if the encryption/decryption logic is in the hardware of the modem (which seems to be the case for some of the beet-kits) you wouldnt need to offload at all...

arpiecodes, 2023-03-09:
Running the serkri distribution sometimes already shows slowdowns when interacting with it through REST (at least, in my case) while not doing much with it, really. I'd not like my charge controller to end up being busy with a lot of extra stuff jammed into it to the extend of that potentially causing issues with the core functionality/compromising it. It would also raise complexity even further.

Besides, we would still need a way to react to stuff like 'identifying' the car, send through SoC and be able to control it from (for example) HA. The official repo doesn't support this as of this moment either.

I'd much rather see something like this added to Sensorbox v3 (and have that assume the master modbus device role, also run the REST API and MQTT client for example) vs. trying to cram it all into one device.

dingo35, 2023-03-09:
Well lets agree to disagree on that.

Mind you that having Sensorbox V3 taking over the modbus master role is a really bad idea, since the Sensorbox is optionally. There are a lot of alternatives out there that can get the Mains currents into the SmartEVSE.

arpiecodes, 2023-03-09:
Agreed with disagreeing. :-)

I was just referring to the Sensorbox as an example. I would also not mind it becoming a totally separate module.

In the near future not only your car will be talking over Homeplug GreenPHY; it will most probably become the basis for smart grid communication. So it can potentially also talk with (for example) solar inverters and other appliances like heatpumps, etc. Would be too bad if the module could not support such nice use cases. While I do agree that's out of scope (like you said with the alternatives that are there for main metering), why not consider it from the start? It would give so much more flexibility vs. making it a single-purpose solution.

I also don't see why having Sensorbox as the modbus master would be such a bad idea. It has been discussed here multiple times already. Nobody loses anything w/regards to functionality vs. current situation if you make it opt-in. If done in a correct way, would only mean the SmartEVSE can act as a Modbus slave which opens up a whole lot of extra possibilities in different set-ups as well. And would allow the SmartEVSE just to focus on that one thing it does best; be a charge controller without any added fuss.

Now, creating a whole different version of SmartEVSE as a whole sounds like complete overkill and would only make firmware management/updates more difficult, require everyone to replace their existing units, etc.

dingo35, 2023-03-09:
You will always need the possibility to have SmartEVSE working as modbus master, so having it also making it work as a slave increases complexity (both in code and in configuration).
Also it goes against the modbus architecture; all communication has to be initiated by the master by design. Almost all use cases are controllers (master) reading data from sensors (as slaves).

Id rather solve the structural problem of modbus only allowing one master by having the SmartEVSE functioning as a flexible configurable gateway from modbus to REST and/or MQTT, so it can interface to HomeAssistant or other more high-level domotica systems.

Just my 2 cents...

arpiecodes, 2023-03-09:
> Id rather solve the structural problem of modbus only allowing one master by having the SmartEVSE functioning as a flexible configurable gateway from modbus to REST and/or MQTT, so it can interface to HomeAssistant or other more high-level domotica systems.

Which is in itself definitely not a bad idea/solution. I definitely support the motion to add more connectivity (e.g. REST API/MQTT, etc) to the SmartEVSE.

I just don't agree with the part where we'd add even more responsibilities/logic to the SmartEVSE's core code; including a whole new networking stack required to make GreenPHY work.

arpiecodes, 2023-04-05:
> @arpiecodes I'm pretty interested in what CODICO has to say. Thanks for the link though, I hadn't spot the carrier boards before. I ordered one, will update you all. Like you said, pretty awesome to [...]

Hey @ArendJanKramer, were you able to try it out yet? I actually never heard back from CODICO and considering buying the carrier board anyways. Thought it'd be good to reach out to check how you are fairing with it first.

ArendJanKramer, 2023-05-04:
@arpiecodes I received the board. Wasn't exactly easy, they asked a lot of questions before they sent one over. Quality is really good, I'm impressed. Still working on getting a Raspberry PI, so I can actually start developing with it.

fluppie, 2023-05-07:
@ArendJanKramer 
Since I'm fan of having this functionality developed, could you work with a Raspberry Pi Zero 2W? Or do you need a 3B or 4?
I can sell you one ( for normal MSRP prices :) ) since I migrated most of my stuff to a LENOVO ThinkCentre M73 (see eBay, you can get them at 75-125 euros) with Proxmox VE on it. I can easily run 4-5 VM's with Debian etc on these machines with an i7 and 8GB of RAM.

ArendJanKramer, 2023-05-08:
@fluppie Thanks for the offer, I think I might be able to get one freed up in my setup.  The Lenovo is actually a pretty good idea though lol.

Another question to the experts here is about zero-crossing and connecting the mains.
See picture of board without the lid:

[image]
[image]

You see "YT-35636 (292877)" modules used for [AC_L and AC_N] and [CP + PE].
The CP + PE pair makes sense for me, but I'm not so familiar for AC lines, is it needed for basic communication?
Because in fact, DC chargers are not using these pins am I right? 

 1 - https://www.codico.com/en/mpattachment/file/download/id/831/ section 5:
"Zero-cross detection has to be used in all applications with communication over AC mains or when module's
power supply is powers from AC mains. Two or more HomePlug AV/GreenPHY logical networks can only coexist
when zero-cross detection circuit is implemented.". In other words, not needed if only one car in the network right?
 2 - The dev board has the proposed schematic in place though. Do I need it? Which phase do I connect here? Does it need to be the same phase as AC_L? 
 3 - According to components that I see, I can just connect SmartEVSE and this board to the same CP signal. I didn't do the math, but since only high frequent components would pass the capacitor on the dev board, the smartevse would still be able to see proper line voltages to negotiate charging with car. Likewise with duty cycle signal. Anyways, the worst that can happen is not a lot so I'll just give it a try I guess.

Again, I'm completely new to this ISO, but I'll learn as we go. "what can possibly go wrong"

fluppie, 2023-05-08:
I would also think CP + PE is the way to go. Since the AC lines are disconnected by the contactor before charging.
Regarding connecting the CP in parallel, I think that's maybe a question for @mstegen ?
Maybe also check this document: https://github.com/uhi22/pyPLC/blob/master/doc/EvseMode.md

>
> The CP of the vehicle is connected to the hot side of the homeplug modem and the hot side of the arduino-charging-logic. The PP of the car is connected to PE via 1k5 (inside the plug). 
>

I also saw the earlier posted link to https://github.com/uhi22/pyPLC#example-flow which looks like interesting information

ArendJanKramer, 2023-05-08:
@fluppie Great resources, thanks for them. That seems hopeful.

I also read this thesis: https://ltu.diva-portal.org/smash/get/diva2:1546405/FULLTEXT01.pdf
Which is great for starters, but lacks some details, specifically on the AC side. 

Regarding the 5% duty cycle to initiate communication, I see this in the code:
https://github.com/SmartEVSE/SmartEVSE-3/blob/main/SmartEVSE-3/src/evse.cpp#L557
When idle, there's no PWM signal. Perhaps by setting that to 5% duty cycle gets things running just as a test.

Maybe @mstegen can indeed confirm if it is safe to connect the dev board in parallel on the CP signal.

mstegen, 2023-05-08:
Great that you received your dev board.
I think you can connect the AC_L and AC_N to L and N of the SmartEVSE.
As the homeplug data is injected on the CP signal, you can connect the CP directly to the CP of the controller.
Ideally the EVSE should have a filter to suppress the homeplug HF signal. 
(it's described in the ISO-15118_3 standard)  

If you kill it somehow, let me know, and i'll replace it.

Ref the 5% PWM, this is required to start digital communication with the EV. 
There is no harm in setting PWM to 5%, If the EV does not support it, it will reject it.
It has been part of the old J1772 standard, so even cars from 15+ years old should be able to handle it.

ArendJanKramer, 2023-05-08:
Ok, here's what I got so far.

Modified firmware. Apparently, there's a few seconds delay before state B is switched in SmartEVSE.
I abused this by setting the duty cycle to 5% over there, so the car has 5 seconds to enable communications.
[image]

I also did some janky wiring, it's "temporary" as we engineers tend to call bad work.
[image]

The funny thing is, it seems to kinda work. When I run "evse" command, part of open-plc-utils I see slac messages. Not exactly the right thing perhaps, but some things are going on:
[image]
These messages definitely stop when I stop the charging.

But next, HOW THE HELL DOES THIS WORK?
I can't find a list of commands or parameters to send / receive so we can find SoC and other stuff. I'm a bit clueless, did some googling but I'm missing the right words apparently.

arpiecodes, 2023-05-08:
@ArendJanKramer Nice progress! Great to see the communication is established.

For me, the following really helped with getting a general understanding on what happens in the digital communication; https://github.com/uhi22/pyPLC#example-flow. If I interpret your results so far, I'd say the evse (charger) has to send SLAC_PARAM.CNF. It is actually a VERY expansive process it has to go through.

It seems someone has managed to (sort of) start a session with that pyPLC stuff here; https://openinverter.org/forum/viewtopic.php?p=56188&sid=9579fd29268d0e332c81ed528f59f09b#p56188. Maybe it helps.

fluppie, 2023-05-08:
I've came across these two repo's who contain a lot of info, especially the one from EDF seems to have a GUI.
https://github.com/EDF-Lab/eVDriveFlow

and also this one: https://github.com/SwitchEV/josev

dingo35, 2023-05-09:
Are you sure your ipv6 stack is in order on your Pi? In that openinverter topic https://openinverter.org/forum/viewtopic.php?t=2262&start=100 they have the same problem, and they attribute it to the software using old-school ifconfig instead of ip route. 
So it might be that your pi does not have a link-local ipv6 address which prevents the communication going forward...

ArendJanKramer, 2023-05-09:
@dingo35 I guess it does? Not the expert, but good suggestion.
I got iso15118 python running (from SwitchEV), here's what we see:
[image]

And this multicast ip you see at the bottom does ping:
[image]

Which is in fact my eth1 (SPI Red Beet) ipv6 address. 

Next to try is pyPLC from fluppie, but since I installed minimal Raspbian without X, there's no window server to run tkinter (which it the code uses all over the place). Will install window server / strip out tkinter out of the source to get things rolling. 
And if not, I'll try to understand the flow more to see if we can get at least 1 step further than I have now, which enables the path for the subsequent steps I guess.

mstegen, 2023-05-09:
Looks like EVerest has everything figured out already. Their software stack is fully open source, and very advanced.
https://github.com/EVerest

See also https://www.youtube.com/watch?v=OJ6kjHRPkyY

ArendJanKramer, 2023-05-09:
@mstegen I've seen that one, but unfortunately it's quite hard to build.
It's basically a mix of all libraries mentioned out here, it pulls in Steve, iso5118 lib, etc.
Will definitely give it another try, but for now I got stuck at the lack of Rust compiler and clang12 installation.
Went all the way through the workspace process, but it went in too many errors for now. Once I have some of the individual components working, I'll might give it a bit more time.

However, I got 1 step further with pyPLC.

Modified code quite a bit to sniff on eth1, and run without GUI.
In the end it seems to take steps until Checkpoint102.

Now, the car is 100% charged and car doesn't seem to execute charging procedures anymore.
[image]

Not exactly sure what ConnectionLevel means in this context, need to read more in the code to learn I guess.

Anyways, enough for today. Tomorrow I'll play again and see if we can get at least one step further.
Perhaps I might also connect a scope to verify there's actually the signals on the CP that I expect.

arpiecodes, 2023-05-09:
I think you are running in PEV mode, which is actually the vehicle side. You should run in EVSE mode for it to act like a charge controller.

ArendJanKramer, 2023-05-09:
@arpiecodes I think there are some things mixed up here, in the debugger I clearly see the evse sim started. Both ini file an py file are changed to start evse, but you are right in that the events are reversed. I’ll dig deeper tomorrow. It might have been a bit too long day for me to notice it’s still wrong around.
Filename and print at top are static btw, I will update them as well.

dingo35, 2023-05-10:
@ArendJanKramer , on a raspberrypi 4, they forgot the dependency on the rust compiler; you need to do a apt install rustc .
I did not notice missing dependencies with clang12...

ArendJanKramer, 2023-05-10:
First things first, check if what I thought I made actually was that I made.
So I hooked up my scope to the CP pin. Guess what, not 5% duty cycle but 100%; whoops.
As a quick check commented out all calls in EVSE code and hard-coded 5% in.

Immediately pyPLC saw traffic coming in on server at port 15118:
[image]

On the scope, the duty cycle is near 5% (ignore probe settings and voltages, lazy is my second name):
[image]

Seems pretty nice to me, very cool to see these messages.

Turns out my modified code was actually doing what I thought it did. Thanks for letting me doubt myself @arpiecodes, actually learned some new things while stepping through the code again. You made me find a hidden test in pyPlcTcpSocket.py, which is actually pretty nice to know about.

Sorry for my spam again, but I'm glad to be 1 step closer. Next is to making sure I can still charge, while also having 5% when I need to set up communication. Will need to familiarize myself with the SmartEVSE code before I can add some REST calls to post the SoC and request 5% PWM.

fluppie, 2023-05-10:
Cool! I see 3 SoC's, FullSOC, BulkSOC,EVRESSSOC
94, 84 and 56 is by accident any of those 3 numbers the right SoC? :).

arpiecodes, 2023-05-10:
@ArendJanKramer That is some amazing progress!

I am curious whether the car would still allow digital communication/give updates on SOC while charging through the 'old' method (e.g. SmartEVSE controlling the charge session instead of through PLC). That would be ideal.

Else, it'd mean we'd have to completely implement charge control through EXI/PLC. Which would most probably almost completely mean rewriting something akin to the core logic in SmartEVSE (and also making the logic kind of redundant there).

ArendJanKramer, 2023-05-10:
@fluppie Those seem to be the right numbers, which make me quite happy.
@arpiecodes This is what I'm working on. Made some buttons in the SmartEVSE page to overrule PWM so I can set up communication and then resume normal charging. The code is new to me, and I have never used the debugger functions before, so need to learn a bit. I will update you once I know a bit more.

ArendJanKramer, 2023-05-11:
@arpiecodes I'm afraid the answer is no, at least for my car.
I just played for a short bit with SiwtchEV's josev, and got it working.

The steps I took:
 - Edit .env in iso15118 to match setup.
 - Run iso15118/secc/main.py. You'll see the tcp server waiting for communication from car
 - Now, you need slac to initiate communication.
 - Edit pyslac/examples/cs_configuration.json, to only have 1 evse
 - Set PWM to 0% in EVSE, I made some buttons to help me with that
 - Run pyslac/examples/single_slac_session.py
 - Set PWM to 5%, immediately you should see pyslac negotiating with car
 - Few seconds later, you'll see iso15118 talking with the car
Much more mature than pyPLC. Perhaps I didn't take the best route in hind sight, but I learnt a lot which is something I don't regret.

**SmartEVSE side PWM control**
[image]
I abstracted the CP signal calls into a function (SetPWM) and I have a global variable with OverridePWM and its value. Does the job for me. If I want to set up a new connection, click zero and then few seconds later at 5. 25% and 100% are unnecessary in hindsight. Reset removes override so I can charge normally. 

**Limitation of car**
I found a few things. First, my car only supports DIN_SPEC_70121. Unfortunately, this only supports the transfer mode of DC:
[image]

**Attempt 1: Disable DIN_SPEC_70121 from SECC side.**
Just remove this option from the list of protocols communicated with car, so an alternative is hopefully chosen.
[image]
RIP, car responds it can't do that, 100% expected of course, but can't say I haven't tried.

**Attempt 2: Ignore spec, hard code AC response in DIN_SPEC_70121**
Basically commenting out the check in fist image. 

[image]

Car rejects this, and says it's not compatible with my charger.

**Conclusion**
My car is not suitable for communication + charging via AC simultaneously, since it only supports the DIN spec, which only support transfer mode of DC. Other cars do support newer protocols.
Is it a failure for me? Absolutely not, because I can extract SoC before charging, and I know the battery capacity. By using the EV meter, I know exactly how much energy has been gone into the car, so I can compute the new SoC quite reliable I think. 

Next steps: Add some endpoint and rest callbacks to SmartEVSE, so when the cable gets plugged in, the digital communication is initialized by the PI + Modem. Then, I can just post the SoC values to SmartEVSE and we can do the energy counting from there, if EV meter is connected.
A more mature approach would be to solve this over Modbus and make the iso15118 more self contained. It depends a bit on where we want to take this feature.

arpiecodes, 2023-05-11:
@ArendJanKramer sad to read about those limitations. But as you already found; there is a workaround that could definitely work for your use-case.

I suppose I also should order a board soon. I am eager to try it out on my Polestar 2, which IIRC does not really have support for Charge 'n Go (which is actually build upon the PLC stuff) so my mileage will probably also vary. But if I can get a SOC and have the car make itself known, then yes, it would also be a full win for me.

What car did you test it with?

dingo35, 2023-05-11:
@ArendJanKramer If you have a diff for me, I can integrate it in the github repo so it will save others the work of fiddling out the 5% pwm stuff...

dingo35, 2023-05-11:
I ordered the poor man's hardware...
[image]

....but now awaiting firmware update of my car, in week 20.

dingo35, 2023-05-12:
@ArendJanKramer thx for the PR, working on it, I take the liberty of taking out the 25% and 100% pwm buttons since you say they turn out to be of no use.
Also question, I want to resolve this compiler warning but don't want to guess:
src/evse.cpp:2040:70: warning: suggest parentheses around '&&' within '||' [-Wparentheses]
                         if ((State == STATE_B) || (State == STATE_C) && !CPDutyOverride) SetCurrent(Balanced[0]); // set PWM output for Master

What would be the correct placement of the parentheses?

dingo35, 2023-05-12:
Notice that when with Arend Jan's code you set the PWM to 0%, the pilot line is no longer kept at 12V and the SmartEVSE will report a vehicle is connected, even if it is not :-)

ArendJanKramer, 2023-05-12:
@dingo35 
Normally, the PWM is updated when in State B or C; so load balancing is updated every cycle.
For override we ignore it, so this would be correct:
 if ((State == STATE_B || State == STATE_C) && !CPDutyOverride) SetCurrent(Balanced[0]); 

> Notice that when with Arend Jan's code you set the PWM to 0%, the pilot line is no longer kept at 12V and the SmartEVSE will report a vehicle is connected, even if it is not :-)
Nice find, never tried without a car connected.

dingo35, 2023-05-12:
Ok if you check out the current master of serkri distribution you will find Arend Jan's code in there, with a few slight modifications:
- introduced new setting (through LCD screen) Modem that defaults to Not Present but can be set to Experiment
- the new stuff is hidden for Modem == Not Present users
- moved /pwm endpoint to /modem/pwm so we can expand experiments;
- moved pwm json element to evse subdivision
- some small layout stuff

@ArendJanKramer I tested, but since you are the only one on the planet currently being able to test this fully.... please let me know if it tests ok so I can put it in the next release.

ArendJanKramer, 2023-05-12:
@dingo35 It seems good, but my car is 100% charged. Will tests again tomorrow with less charge, can always make a new PR with fixes if needed.
[image]

But the rest seems good, it applies the right PWM signals it seems. 
Thanks for your work, feel free to put in your release.

fluppie, 2023-05-16:
@ArendJanKramer did you play some more with the RED beet module :-)? Or I should also buy a module ( https://www.codico.com/de/red-carrier-board-h-1-1 ) and provide remote access?
I have a 2022 BMW iX3, so I don't know if they support 15118 AC "Plug & Charge". I do have the latest 03/2023.45 iDrive software.

arpiecodes, 2023-05-17:
@ArendJanKramer Good idea's! I would definitely be in the market for an easy to use solution without spending 100+ euro's.

How about an alternative solution to the above list; a solution that would act more like a transparent 'bridge' between ethernet and the GreenPHY network? A bit like the powerline adapters that are being modded. Maybe slap on a modbus/serial connection to it as well. In such case, an ESP32 would probably do. A Raspberry Pi would also do the job, but requiring one for it to function would increase the cost dramatically and makes the solution very specific.

I don't think a full linux SOC would be required. Most people using SmartEVSE right now already use e.g. HA or some other platform with a Linux OS. ESP32 would be a cheap option, lots of I/O options for future expansion and in theory would also easily support bridging between the GreenPHY network and any other ethernet network.

ArendJanKramer, 2023-05-17:
> @ArendJanKramer Good idea's! I would definitely be in the market for an easy to use solution without spending 100+ euro's.
> 
> How about an alternative solution to the above list; a solution that would act more like a transparent 'bridge' between ethernet and the GreenPHY network? A bit like the powerline adapters that are being modded. Maybe slap on a modbus/serial connection to it as well. In such case, an ESP32 would probably do. A Raspberry Pi would also do the job, but requiring one for it to function would increase the cost dramatically and makes the solution very specific.
> 
> I don't think a full linux SOC would be required. Most people using SmartEVSE right now already use e.g. HA or some other platform with a Linux OS. ESP32 would be a cheap option, lots of I/O options for future expansion and in theory would also easily support bridging between the GreenPHY network and any other ethernet network.
> 
> Maybe we should explore options to 'team up' with uhi22? His project CCS32 seems to be very alike in goals; https://github.com/uhi22/ccs32. It just misses a good hardware platform to bring it all together, without a hacked PLC modem.

It depends, I'd rather make something and be able to completely forget about it, so I don't have to maintain it. 
On the other hand, if there's almost no demand for this feature, it makes more sense to duct-tape something together with dev-boards + Raspberry's or Powerline adapters. 
Why I mentioned the Linux Soc is because Linux kernel is shipped with QCA7000 series driver and plc utils of Qualcomm/Atheros work straight out of the box,cli utility's just so. You can also run existing Python suites like Josev, EVErest and steve. If you take an ESP32 we'd need to rely heavily on @uhi22 his work and expand that (for example: AC charging). 

I believe regardless of the solution, we need hardware connection between SmartEVSE and module, because SmartEVSE needs to automatically trigger 5% cycle and work with "modem module" to initiate and extract information, issue retries and then start charging. Depending on negotiated protocol SoC can be frequently updated, or it needs to be re-computed based on EVMeter. Using wifi for that feels a bit too fragile, and the last thing I want is a charger that doesn't charge.

dingo35, 2023-05-18:
I agree we need a hard wired connection between modem and SmartEVSE; I see the following possibilities:
-modbus
pro: no soldering, easy access
pro: software library already in use
con: a connection can only be initiated by the master device (the SmartEVSE). I doubt whether the complex communication protocol that has to be followed can be implemented without the modem being able to initiate communication
-uarts:
uart0=debug/program port
uart1= modbus port
uart2: pins seem to be in use for switch and actuator drive
-SPI:
pro: simple and efficient
pro: QCA7000 series support SPI
con: soldering required; is it possible to add another SPI device?
-I2C:
?
-wifi
pro: no soldering required
con: wireless, would only be temporary solution for experiments
-bluetooth
pro: no soldering required
con: wireless, would only be temporary solution for experiments

Perhaps @mstegen could shed some light on this?

uhi22, 2023-05-18:
Hi, thanks for tagging me, I was indeed not aware of this project. Glad to see, there was some success here with pyPLC, and also with Everest. This helps to get the "big picture", thanks for this.

One remark regarding "DIN does not support AC charging": Also in the DIN spec, the AC charging is basically supported, this is visible in the https://github.com/Martin-P/OpenV2G, where we find things like dinEVRequestedEnergyTransferType_AC_three_phase_core. This sounds like that a car which supports only DIN may support also AC charging via high-level-protocol. I have no information, whether there are cars in the field, which actually support this feature.

Remark regarding convincing the Ioniq to use ISO instead of DIN: This is not possible from protocol flow point of view. In the very first messages, the car announces a list of supported protocols. The Ioniq just announces a list with one element: DIN. The charger has only the possiblity to say "I want the n'th entry of the list.", this means, no matter how we patch the charger, the Ioniq will talk either DIN or nothing.

Remark regarging the options to bring ISO15118 together with SmartEVSE: The options are well described above, all have their pro and cons, I see no perfect solution. A "compact, long term solution" would be a re-design, which directly integrates the homeplug stuff into the base unit, e.g. directly driven via SPI from the existing controller. As short-term, more flexible approach, which would be also upgradable at the already installed units, I see the modbus variant. Somehow to make a box ("SECC box") which only needs the CP line, power and modbus. It can permanently listen for SLAC requests, negotiate the SLAC, establish the high-level-communication and provide the result (SOC, success or error code) via a modbus register. The main controller just needs to read it, and can decide how to go on (most likely disconnecting, and providing "normal" PWM for normal AC charging).
The content of such a small box is quite flexible, maybe a raspberry with HomePlug modem (for the self-makers), or a ESP32 with an SPI-connected QCA7xxx. This could be developed independent from the main project.
Even if it comes to real high-level-AC-charging, there is no need for fast communication between the main controller and the SECC box. The time-critical high-level-communication is handled by the SECC box, and on modbus only such things like SOC and status is polled by the main controller, and things like "continue with x amps" are flowing from the main controller to the SECC box. Should be no issue, if this is slow, e.g. once per second.

mstegen, 2023-05-19:
Thanks for your reply.

I noticed on the Red Beet carrier dev board, there is an option to connect the QCA7xxx to mains lines, or you can connect it to the CP line. These are selectable with dip switches. A third connection is used for zero crossing detection. They mention it's mandatory, but on my dev board there's another dipswitch which allows you to switch off zero cross detection. (enabled by default)

Looking at the pictures from @ArendJanKramer, it seems his board had this set as well. 
So knowing this, i am pretty sure he can disconnect the mains connections to the dev-board, and it would have no effect on his experiments..
I'm planning in doing some tests myself today. I've got the dev board communicating with the raspberry pi, with pyPLC in EVSE mode (basically what ArendJan had working last week)

I also have a YellowBeet-E board from 8Devices, they look suitable for the next step in development, as they have the required transformer for interfacing with the CP line (and ZC circuit). 
[image]

If the only requirement is SOC readout, then it might be possible to let a ESP32 handle it. But no clue yet at the moment how to read/write to the red/yellow beet module over SPI, The linux driver seems to convert the SPI connection to a ETH port.

For a future SmartEVSE hardware version, i think it's best to connect it directly to the SPI bus of an ESP32(-S3).
As a first add-on to the current version, it might be an idea to make a raspberry pi shield/hat as ArendJan suggested.

As for communication between Rpi and SmartEVSE, there are no i/o pins left on the ESP, so i suggest using the USB/program port of the SmartEVSE and connected that to the raspberry pi.
State changes, SOC etc, can then be transferred serially.

arpiecodes, 2023-05-19:
Makes sense @mstegen. The YellowBeet-E sounds like a good solution. For me the question was indeed whether the 230v lines were really needed to communicate with the car; if this is/was the case I needed something that contained that extra circuitry for AC communication. If it's not necessary, then for my purposes the Yellow Beet-E will probably do (and combining it with an ESP32). I may also explore options to add it on the Sensorbox.

I am wondering, however, if the CP line communication is available on all cars, or if this differs per brand/make. Of course you will only find out after having received the module and trying that out.

uhi22, 2023-05-19:
At least the Type2 / CCS Type2 cars always use the CP for communication. The use of AC line for communication I only saw in 10 years old presentations. I do not think you will find a car which uses this (at least in Europe).

Regarding "no clue yet at the moment how to read/write to the red/yellow beet module over SPI": I saw a kind of reverse-engineered paper which describes the SPI communciation parameters, but cannot find it anymore. Was from I2SE I think, and they wrote something like "all information in this document was taken from the public linux driver". I understood, that "official" information regarding the protocol will be delivered from Qualcom only if you sign a NDA. But AFAIR the protocol was not too difficult, e.g. for sending an ethernet frame, on SPI some preamble and some short command word needs to be sent, and then the data one-to-one as in a ethernet frame.

uhi22, 2023-05-19:
Found it: Search for "an4 Aurich Wahren" in Google.

[142 more replies]