SmartEVSE/SmartEVSE-3 #26: Relay not closing after terminating the charging process

BitterAndReal, 2022-08-04:
If I'm terminating the charging process (6) and the EV charge controller indicates State B (disconnecting the R2 resistor) the smart EVSE V3 does not open the AC contactor (9b). 
Meanwhile the EV unlocks the plug (9a) and the user can disconnect the plug while there is still power on the plug. 
The AC contactor only opens once I remove AC to the EVSE.
We did not have this issue with V2.2.

[image]

I have not checked what happens when I terminate the charging process from the EVSE.

mstegen, 2022-08-04:
Please check the contactor type, the SmartEVSE v3's SSRs that switch the contactors on/off, don't work well with humm free/ silent contactors. 
Once switched on, they then will then not switch off anymore (although the SmartEVSE has switched off the contactor)

The Iskra IKA432-40 and Hager ESC425 work good with the SmartEVSE.

BitterAndReal, 2022-08-05:
Thanks for your explanation.
We are using quite big 4P 100A contactors.
I guess the quickest solution for now is to place a control relais between the EVSE output and the contactor...

closed 2022-08-05 by BitterAndReal