SmartEVSE/SmartEVSE-3 #27: EVSE Status register

dummyka9609, 2022-08-06:
Hi!
I couldn't find anywhere the meaning of the possible states of the status register (0x0000).
Can anyone tell me where can i find?

bobosch, 2022-08-06:
https://github.com/SmartEVSE/SmartEVSE-2#modbus

dummyka9609, 2022-08-07:
I found that too, thanks. But I don't know what it mean _"A-D (EVSE State), E-H (A-D Waiting for Balance Master)"_
[Here](https://github.com/SmartEVSE/SmartEVSE-3/issues/1) mstegen wrote how I can start the charging via modbus, but I would like to know what each state means exactly.

bobosch, 2022-08-07:
I think the documentation is out of date here. It should return the State as defined in the code:
https://github.com/SmartEVSE/SmartEVSE-3/blob/a91b47bc5032908e46739e2b390bf2171d890706/SmartEVSE-3/include/evse.h#L139-L149

bobosch, 2022-08-08:
Updated README.md (https://github.com/SmartEVSE/SmartEVSE-2/commit/d3263b789483d5297697f0b53ef506fd54d82847)

closed 2022-08-08 by bobosch