SmartEVSE/SmartEVSE-3 #28: New configuration multi evse

ostium, 2022-08-26:
Hello,

I'm trying to use your load balancing "algorithm" on a new project. I want to configure 3 EVSE to work simultaneous connected to same mains. I have also one meter to monitor mains current on 3 phases.

I don't want LCD, buttons, etc... only a basic closed box, that will be configured only by firmware.I need one Master, two slaves and one meter.

Please help me with config. Tell me if this is correct:

EVSE Master Configure:
#define MODE 1 // SMART to read METER
#define LOADBL 1 // EVSE is MASTER 
#define MAINS_METER 1 // I will use SENSORBOX 
#define RFID_READER 0 // No RFID
#define WIFI_MODE 0 // NO WIFI

EVSE SLAVE 1 Configure:
#define MODE 0 // SMART to read METER
#define LOADBL 2 // EVSE is SLAVE and wait for requests/command from master 
#define MAINS_METER 0 // No sensorbox
#define RFID_READER 0 // No RFID
#define WIFI_MODE 0 // NO WIFI

EVSE SLAVE 2 Configure:
#define MODE 0 // SMART to read METER
#define LOADBL 3 // EVSE is SLAVE and wait for requests/command from master 
#define MAINS_METER 0 // No sensorbox
#define RFID_READER 0 // No RFID
#define WIFI_MODE 0 // NO WIFI

SENSORBOX 2 Configure: default

Thanks in advance.

greetings

ostium, 2022-08-29:
Hello again, I'm trying to understand your function "CalcBalancedCurrent" with no success.

As I said above, I'm testing 3 chargers stations powered by same mains. And, if I have a MAX_MAINS of 50Amps and set each charger to MAX_CURRENT of 20A, and I have a baseload (load without any active EVSE) of 40A and connect all 3 EVSE, supposedly, only one of them should turn in state C, because only have 10A available for all 3 EVSE. Right?
But what happens is that when connecting the 3 cars at the same time, they all get 10A assigned by the master. Its, a strange behavior... or, maybe I'm doing something wrong.

Can you explain a little bit about how you get your balanced current? 

Thanks in advance.

greetings

mstegen, 2022-09-12:
Your settings for each controller look ok.

The idea behind the current calculation is that only the -measured current- on the MAINS will determine what is available for a EV that starts charging. 
If you start three EV's at the same time, then this will not work correctly as you found out, as the measured current will not change immediately after the EV starts charging. It can take up to 30-40 seconds before the EV actually starts drawing current. In this time there is still 10A available.

dingo35, 2024-05-31:
Closing topic because of inactivity.

closed 2024-05-31 by dingo35