SmartEVSE/SmartEVSE-3 #29: Crashes when charging Renault Zoe

Xnygra, 2022-08-28:
Having used a SmartEvse V2 for many years, I bought a V3 and replaced my V2. It works on all EV that I tried, but the EVSE crashes when charging Renault Zoe, and the SmartEvse needs to be Power Cycled to work again. 

Having the Mains set to 20A and MAX to 16A. House is typically drawing 5A. 3-phase. When start charge all currents from the sensorbox freezes to a value, typically 9A, and the charing do not start. To get the SmartEvse to work again I have to kill its power and restart it. 

When lower the MAX to 12A, so the EVSE don't need to limit the charge Amps (from the sensorbox readings) it works fine. 

So my workaround is to have a lower MAX charge amp that i would like to have, and again, this only happens to renault Zoe (tested on two vehicles) and not on any vehicles from the VAG group or Hyundai/Kia. Or switch back to the old V2

Im not a coder my self, so I'm a bit lost here, but happy to help with more data/information that might be required to find what's going wrong?

mstegen, 2022-08-29:
Hi, 
What brand of contactor are you using? I've noticed that some contactors will disturb the controller, up to a point where the controller crashes.

Xnygra, 2022-08-29:
I'm using a GE CTX 32A. I got a Schneider sitting at home that i can swap to, to see if that makes any difference.

Xnygra, 2022-09-02:
Swapped to a Schneider contactor, and it seems to work now. Still a bit confused why that solved the problem and why it seems only happens with Zoes.

mstegen, 2022-09-05:
I'm glad you fixed the issue, i agree that it's weird that the GE contactor only with a Zoe causes the issue..

closed 2024-05-24 by mstegen