SmartEVSE/SmartEVSE-3 #30: Protective earth conductivity check

DariusPL, 2022-09-15:
The SmartEVSE v3 does not continue charging once the PE wire is disconnected from the SmartEVSE terminal (CP-PP open loop) but still can operate and continues charging with no PE wire coming from the earthing grid. 
Continuous protective earth conductor presence check - no output when the protective earth conductor opened 
Is this a feature being considered to implement in the future rev?

mstegen, 2022-09-15:
You then need some kind of reference to measure against. Not sure how to implement that yet. 
Some EV's will not start charging if there is a voltage >50Vac between N and PE. It would be a good idea to test for this as well.

Elektrofix-OL, 2023-07-15:
Info: In Germany, a continuity measurement of the protective conductor with at least 200 mA must be carried out, otherwise the measurement is considered not to have been carried out.
The 50V is the maximum permissible touch voltage that may occur.

mstegen, 2023-07-16:
Problem with 200mA is that it will most likely trigger a Residual-current device.

Detecting if the PE is connected to ground can be done with a resistor between PE and L1. If PE is grounded then there will be current flowing through the resistor, which can trigger a optocoupler.
Something like this:

[image]

@DariusPL do you have any idea how the car injects these pulses between PE and N?

Elektrofix-OL, 2023-07-16:
A residual current circuit breaker of 30mA may trigger from 15mA and must have triggered up to 30mA.
A measurement with a current of 200mA is therefore not possible and currents of <15mA are also critical if leakage currents from other circuits are already present.

But does that have to be the case at all?
The charging station must be checked during installation anyway. An impedance test between L-PE is mandatory. The PE should not change if the house installation is intact

The ZOE measures the voltage between N+PE. If the connection is bad, the voltage between N+PE will increase during charging, but this only works for vehicles with asynchronous chargers. This measurement would not work for a Smart with a 22kWh charger, since it has a synchronous charger.

A voltage measurement between N+PE with a high-impedance measuring device could work. However, the current should not reach 15mA.

mstegen, 2024-05-31:
Thanks for your explanation, closing this now.

closed 2024-05-31 by mstegen