SmartEVSE/SmartEVSE-3 #34: Charging stops after some time

jeroenrnl, 2022-10-31:
I have just converted my EVBOX charger to SmartEVSE, but I'm experiencing some issues. The charging stops after some time (I've seen 8 minutes, but I've also seen > 1hr).

I'm reading the status of the SmartEVSE via websockets and NodeRed. When still on my desk, I noticed a couple of reboots whenever I disconnected the websockets, so that may be a cause. The reset reason is 4, I believe that is a watchdog timeout.

Is there a way to remotely monitor the full logs? Leaving a laptop connected for a long time is not practical since the charger is (of course) outside.

Is there a better way to monitor the status of the SmartEVSE?

Is MQTT support planned?

mstegen, 2022-10-31:
The ESP's websockets implementation is buggy, unfortunately. There is a fork of the code that removed the websockets entirely:
https://github.com/serkri/SmartEVSE-3
If i'm not mistaken it also supports MQTT

I'll probably do the same in a future fw update.

jeroenrnl, 2022-10-31:
I have switched to that firmware and it looks promising so far! Thanks for your quick response!

(It doesn't support MQTT, BTW)

jeroenrnl, 2022-11-02:
Unfortunately, that response was too quick... only 10 minutes after that comment, it stopped charging :-(
I have switched to a newer release of serkri's firmware (came out yesterday), but again, it stopped after only a few minutes of charging.
I compiled a debug version of that firmware, and I'm charging now with a debug session open:

(setState)(C0) 23:59:18 STATE B -> B1
(ProximityPin)(C0) PP pin: 273 (597 mV)
(EVSEStates)(C1) Cable limit: 32A  Max: 16A
(setState)(C1) 00:00:00 STATE B1 -> B
(EVSEStates)(C0) Diode OK
(setState)(C0) 00:00:30 STATE B -> Activate
(setState)(C0) 00:00:33 STATE Activate -> B
(EVSEStates)(C1) Diode OK
(setState)(C0) 00:02:46 STATE B -> A
(ProximityPin)(C0) PP pin: 273 (597 mV)
(EVSEStates)(C1) Cable limit: 32A  Max: 16A
(setState)(C1) 00:02:49 STATE A -> B
(EVSEStates)(C1) Diode OK
(setState)(C0) 00:02:50 STATE B -> C
(setState)(C0) 01:07:16 STATE C -> B
(EVSEStates)(C0) Diode OK

It charged for a little over an hour and then stopped. The car indicates it is an unexpected stop of charging. Any ideas?

mstegen, 2022-11-02:
It looks like the controller determined that the EV changed the state from C to B, except this was not the case.
The Pilot() function determines the min and max measured pilot levels, and uses that to return a 6V(state C) or 9V(state B) level.

In my beta version i have since changed the levels to:
if (Min >= 3055 ) return PILOT_12V;                                     // Pilot at 12V (min 11.0V)
if ((Min >= 2735) && (Max < 3055)) return PILOT_9V;                     // Pilot at 9V
if ((Min >= 2400) && (Max < 2735)) return PILOT_6V;                     // Pilot at 6V

Can you try to change these, and see if it makes any difference?

jeroenrnl, 2022-11-02:
Unfortunately, same issue, this time after only 1 minute of charging...
```
(setState)(C0) 23:14:59 STATE B -> C
(setState)(C0) 23:16:09 STATE C -> B
(EVSEStates)(C0) Diode OK
```

jeroenrnl, 2022-11-02:
I've added some extra debugging to the code:
```
    for (n=0 ; n<25 ;n++) {
        sample = ADCsamples[n];
        voltage = esp_adc_cal_raw_to_voltage( sample, adc_chars_CP);        // convert adc reading to voltage
        if (voltage < Min) Min = voltage;                                   // store lowest value
        if (voltage > Max) Max = voltage;                                   // store highest value
        _Serialprintf("PP pin: %u (%u mV)\n", sample, voltage);
    }
    _Serialprintf("min:%u max:%u\n",Min ,Max);
```
This generated about half a million log lines in a few minutes :smile:

I was able to run a test twice, thanks to the fact that it stopped within a minute. In both cases, I see the PP rise before the disconnect. That would indicate the car initiating a stop, right? I also noticed that now the car is fuller (not yet at the charge limit), it stops within a few minutes... could it be that the car is signalling a lower charge current?

```
(Pilot)(C1) min:2471 max:2484
(Pilot)(C1) min:2477 max:2484
(Pilot)(C1) min:2477 max:2874
(Pilot)(C1) min:2860 max:2878
(setState)(C1) 00:06:06 STATE C -> B
(Pilot)(C1) min:2474 max:2484
(Pilot)(C1) min:2474 max:2481
(Pilot)(C1) min:2474 max:2860
(Pilot)(C1) min:2853 max:2868
(setState)(C1) 00:07:15 STATE C -> B
```

mstegen, 2022-11-03:
Thanks for the debug info.
The only way a car should signal that it's fully charged, is by changing the state from C to B. 
To the ESVE your measurements looks exactly that. Somehow it's measuring >25 samples that are in the 9V range, that can not be noise anymore.

So i think the trigger for the car to stop charging must be caused by something else.
Can you please measure the voltage between PE and N while it's charging?  Some EV's will monitor the ground(PE) connection, and stop charging when they detect a voltage >50V between PE and N.

jeroenrnl, 2022-11-03:
The car is reporting a voltage drop and also refuses to charge again until I unplug and replug.

The weird thing is, that I have not touched the high voltage part at all. This is just a few years old EVBOX that Eneco refuses to support, so I took out their control hardware and put in this + an ESP32 to monitor to S0 bus.

I don't really have the means to do a proper measurement, I have a cheap Chinese multimeter that I don't really trust with AC voltages (and rightfully so... I just measured my outlet here and it tells met it's 140mV DC :sob: ) and although I should have an old Fluke laying around somewhere, it's still not really practical to long term measure with a handheld multimeter, very easy to miss something, but I'll give it a try. (when it's not raining... I don't like the idea of poking in a live circuit in the rain :flushed: ). Any other ideas on how to do such a measurement?

Another thing I was thinking is to temporary remove this from the code and see if it continues charging... since the EVSE is only switching power off and on and doesn't do any current limitation or whatsoever, this should be safe to do right?

jeroenrnl, 2022-11-05:
I have done some measurements and usually see around 300mV between ground and neutral while charging.

But... I have now seen what the issue is. Prior to the car giving up on charging, the high voltage switch (HS40-40) turns on and off rapidly. I measured the voltage betweem N and C1 (A1/A2) and as long as charging was happening, it was around 230V, but it dropped to 0V, I didn't see the fluctuation there, but that might have been the multimeter not keeping up with the rapid changes.

When the switching starts, the car reports varying voltages, probably because it does an average over a few measurements and at some point it gives up, with an error voltage drop. 

Sometimes this happens within a minute after charging started, but I've also had a few charge sessions that lasted half an hour or more... I've also seen one time where the rapid switching stopped before the car stopped the charging and then it just continued as normal.

I have checked the code and see no other point where it switches the C1 connector, besides when changing from State C to State B, so it doesn't seem like the switching is caused by the code. What else could be causing it?

I am including my block diagram / wiring diagram, maybe there's a hint there?

[image]

mstegen, 2022-11-05:
Thanks for the detailed wiring diagram. 

The erratic behavior of the contactor is clearly not good. Maybe the first SSR output (C1) of the SmartEVSE is not working properly. As a test you can try to use the C2 output. In the standard software it will switch both outputs simultaneously. 

The extra ESP32, if powered with a LDO can cause some issues.
If the circuit does use a DC-DC converter, it should be ok. The 12V output can deliver max 200mA.

jeroenrnl, 2022-11-05:
It's powered by a small step down converter. I think MP1584, but not sure. I will check tomorrow.

I will also see if it will be possible to switch to the C2 connection, mostly depending on weather.

jeroenrnl, 2022-11-06:
It looks like we've found the issue. I changed the wire from C1 to C2 and swapped the pins for C1 and C2 in `evse.h` (not sure if that was necessary, but did it just to be sure, I'm still on serkri's firmware). This morning I had a successful charge of about 30 minutes (first time it actually completed a charge) and then, after a ride of a couple of 100km's, I charged it again this evening and it charged for about 4 hours without an issue.

I'm not sure if I need the car much this week, so I'm not sure if I'll be able to do any further tests later this week, I'd also would like to do a verification, back on C1, just to make sure it wasn't a wire that wasn't tightened properly or something, but my battery is now at 90%, so I will need to drive a bit before I can add more.

mstegen, 2022-11-07:
Good news! 
Take your time, If you come to the conclusion that the C1 output is not working properly, please let me know, then we'll swap the controller for another one.

jeroenrnl, 2023-10-04:
This was solved (a long time ago) by replacing the unit.

closed 2023-10-04 by jeroenrnl