SmartEVSE/SmartEVSE-3 #38: Question related modbus/p1 and

nekocentral, 2023-01-23:
Currently i'm using a E20 p1 from zuidewijk. Currently i'm planning getting ethernet wired to the garage as i don't have wifi there.
I'm wondering if there is a way to hardwire evse. if that is not possible, i wonder if i can get the P1 data to it without a sensor box mabye a hardwired esp or arduino to modbus. 
Do you have any tips on doing that or hints on what to use?

nekocentral, 2023-01-23:
Else, i know it can speak to modbus energy meters, can i use it with a Shelly Pro 3EM?

mstegen, 2023-01-30:
It's not possible to connect ethernet directly to the SmartEVSE. It does have a wired RS485 port, and Wifi. 
But if you are planning on getting wired ethernet to the garage, why not add another ethernet cable, and use that as RS485 bus?

The SmartEVSE will also work with other modbus energy meters. Alternatively you can install the Sekri fork. That has support for   other ways to get the P1 data to the SmartEVSE.

mstegen, 2023-07-26:
This is currently not possible. The Serkri fork (https://github.com/serkri/SmartEVSE-3) has support for Modbus TCP, but as far as i know only to read/write the controllers modbus registers.

dingo35, 2024-05-31:
The modbus TCP bridge had to be removed because it disturbed the "regular" traffic on the modbus. We will look into making more registers of the MainsMeter and the EVMeter available, if it is possible without bandwidth problems. For other devices (that are not SmartEVSE related, like Solar/PV Meters or other stuff), we recommend you operate them on a separate modbus with a USB<->RS485 adapter, so your Domotica software can be the "master" of those devices.

closed 2024-05-31 by dingo35