SmartEVSE/SmartEVSE-3 #41: Which pins show that car/cable is connected?

leadenchopsticks, 2023-04-02:
I have connected the SWITCH terminal of my SmartEVSE3 to a switch actuator of my KNX installation, so I can trigger the EVSE on/off based on the lowest hourly electricity rate. This is working great. Now to further optimize my scripts I'd like to know if my car is connected to the EVSE, even when not charging (e.g. switch is off). I was thinking of using the locking actuator pins for this and connect them to a binary input device. However I guess that the locking actuator will not do anything when I connect my car while the EVSE is still off.

Would this be possible at all using the SmartEVSE pins, or should I be looking at using an API for this?

arpiecodes, 2023-05-25:
You could take a look at the Serkri fork, the version there offers an HTTP API you could use to determine whether the car is connected or not. https://github.com/serkri/SmartEVSE-3

dingo35, 2024-05-31:
You can use the HTTP REST API, or the MQTT API, or the HomeAssistant integration for this; look at the docs!

closed 2024-05-31 by dingo35