SmartEVSE/SmartEVSE-3 #43: Menu doesn't work

WanderW, 2023-04-14:
On a (new, out of the box) SmartEVSE with firmware 3.0.1, the menu doesn't work.

Chrome devtools shows:
"caught ReferenceError: openNav is not defined  at HTMLSpanElement.onclick ((index):28:92)"

The function "openNav()" is defined in script.js, but that file isn't loaded from index.html

mstegen, 2023-04-14:
If you mean the web page menu, yes the "hamburger" does not work yet.
Currently it's only a status page that shows the Phase currents , Charging state and Errors (if any)

rob040, 2024-02-13:
+1

> function "openNav()" is defined in script.js, but that file isn't loaded from index.html

and script.js is present in spiffs, so simply adding 1 line to index.html would solve that:
  `<script src="script.js" type="text/javascript"></script>`

However, I guess the subsequent js/html code isn't production ready yet...

Are there any prognoses on issue solving / further development ?
or should we all refer to the serkri fork ?

mstegen, 2024-02-13:
Yes, dingo35 is creating a stable 3.5 version, that will have a working webpage.
https://github.com/dingo35/SmartEVSE-3.5

Once that's stable, i'll program the controllers with this software by default.

mstegen, 2024-05-24:
From now on i'll program controllers with v3.6.0. You can of course upgrade the fw of your controller to this version.

closed 2024-05-24 by mstegen