SmartEVSE/SmartEVSE-3 #44: 6A minimum charge current per phase

PETVK, 2023-06-05:
Hi,

The minimum charge current for the car per phase cant be set lower then 6A. My car (MG4) accepts 6A total (3 phase), and 2A per phase.
To charge my car purely on solar energy it would be beneficial if the setting could be set at 2A

Thanks in advance,

Peter

mstegen, 2023-06-05:
Unfortunately the standard does not allow the charge current to be set lower then 6A.

arpiecodes, 2023-06-05:
If you goal is to consume all your solar generated power 'as seen from the meter' (so not the 'actual' power), then you could theoretically switch over to single phase charging and apply a 6A current limit. It does require some extra steps and considerations, though. May want to check out the serkri fork (https://github.com/serkri/SmartEVSE-3) which has more features and also has a description on how to implement the single/three phase differentiation.

PETVK, 2023-06-06:
Its a bummer that the standard doesn't allow lower than 6A but the proposal of arpiecodes fixes the issue, **thanks!**

uhi22, 2023-06-06:
Just to make it more clear: The solution still is according to the 6A minimum current. It does not go below the 6A. The "trick" is just to have 1 phase with 6A, which is 230V*6A = 1380 Watts, instead of the 3-phases which are 3 * 230V * 6A = 4.1kW. Did I understand this right? Or is there (for certain cars) the possibility, to apply a non-standard PWM *below* 6A, to convince it to draw less current?

arpiecodes, 2023-06-06:
> Just to make it more clear: The solution still is according to the 6A minimum current. It does not go below the 6A. The "trick" is just to have 1 phase with 6A, which is 230V*6A = 1380 Watts, [...]

That's correct. The car only has one amperage limit and it applies to all phases in use. So the trick would effectively limit the current to 6A when you only 'give' the car single phase power. This can be achieved by using two contactors; one 4p 'main' contactor  and another 2p contactor to switch the extra phases on/off. Of course switching has to be done with care (and to prevent unsafe situations, the 2p contactor is chained behind the 4p one).

The standard does not allow going below 6A for current limiting.

dingo35, 2024-05-31:
There are some users (if I remember correctly especially Tesla-users) that have been able to lower the minimum charging current to 3A; we would like to stick to the standards, BUT I have made a compiling option, so people can make their own firmware.bin on their own risk: if you compile the firmware with option -DMIN_CURRENT=X, X will be the MinCurrent value (in Ampères) that will be used in the firmware.

closed 2024-05-31 by dingo35

bobosch, 2024-06-02:
I can reduce the charging current with the Tesla App to 5 A ... but not with the PWM signal :-(

dingo35, 2024-06-02:
https://github.com/dingo35/SmartEVSE-3.5/issues/51