SmartEVSE/SmartEVSE-3 #48: Connection to SolaEdge

devdems, 2023-09-12:
Is there any documentation available regarding the connection of the charger to SolarEdge?

I have InePro as an Import/Export meter connected to RS485-1. How do I get data also to the charger? Do I connect to the same RS485-1, do I connect it to RS485-2 or does it get data from a TCP/IP connection with SolarEdge?

Is there any community group for this charger as asking this info in the issue section is not ok.

mstegen, 2023-09-12:
I looked at the InePro380 datasheet, and it should be possible to use it as a Mains meter for use with the SmartEVSE.
Problem is that with modbus, there should be only one modbus master. So you can not connect it to the Solaredge and SmartEVSE at the same time.
The modbus registers can be configured by selecting the custom modbus meter type. The baudrate of the meter should be set to 9600 bps, 8 bits no parity. you can set the address of the meter to any address above 10.

devdems, 2023-09-12:
@mstegen thank you for the reply. I need to have a connection from the meter to SolarEdge. What about getting this info from SolarEdge and not directly from the meter?

closed 2024-04-09 by devdems

TRIROG, 2025-05-03:
> [@mstegen](https://github.com/mstegen) thank you for the reply. I need to have a connection from the meter to SolarEdge. What about getting this info from SolarEdge and not directly from the meter?
> 
> EDIT: Based on the bottom code I guess SolarEdge is already supported and the charger can be connected to the other port RS485-2 on SolarEdge which needs to be set as SunSpec (Non-SE Logger). Am I reading this correctly?
> 
> #define EM_SOLAREDGE 6
> 
> {"SolarEdge", ENDIANESS_HBF_HWF, 3, MB_DATATYPE_INT16, 40196, 0, 40191, 0, 40083, 0, 40226, 3}, // SolarEdge SunSpec (0.01V (16bit) / 0.1A (16bit) / 1W (16bit) / 1 Wh (32bit))

@devdems did you figure this out? Does it work as described i.e. that smartEVSE can be connected to RS485-2 port on SE inverter and read energy meter data connected to port RS485-1 on SE inverter?