SmartEVSE/SmartEVSE-3 #49: sdp request

Fatih738, 2023-09-14:
I have codico red carrier board. I have used stm32f4 to communicate with qca chip. I have used https://github.com/uhi22/ccs32clara. Can I make a ccs charge controller with this code. I know that I have to implement cp duty cycle and pp reading. How can I test stm32f4 and red carrier board as a pev side.
Also I have white carrier board EI. I have ran Seventax FreeV2G code. I have connected CP and PP pins to red carrier board. But I havent set connection between these modules. I have raspberry pi, 3 red carrier board, white carrier board EI. CAn you help me about testing ccs charge controller with these board? How can I test these boards

SLAC matching succesfull. Im using white carrierd board EI with freeV2G as evse side. stm32 and red carrier board as pev side.
/* our link-local IPv6 address. Todo: do we need to calculate this from the MAC? Or just use a "random"? /
/ For the moment, just use the address from the Win10 notebook, and change the last byte from 0x0e to 0x1e.
You wrote these comments in code (ipv6.c). Idont understand here. Can you explain
Also after slac matching succesfull. [SDP] initiating SDP request. I dont pass the [SDP] initiating SDP request. How can I do sdp

thanks

Fatih738, 2023-09-14:
[image]

I have received these outputs over uart

thanks

Fatih738, 2023-09-14:
[image]

uhi22, 2023-09-14:
Seems the codico is crashing. You could try to use pyPLC as evse, this would make it possible to record a pcap trace, which would help analyzing the topic.

Fatih738, 2023-09-18:
Can I run pyPLC on raspberry pi with codico(red-carrier-board-p)?

uhi22, 2023-09-18:
Yes, the pyPLC runs on Raspberry, and according to the datasheet of the redbeet, it is compatible with the QCA linux driver. I never used this combination, but others did successfully.

Fatih738, 2023-09-18:
Can I coontrol all charge process with pyPlc as pev side on raspberr pi

uhi22, 2023-09-18:
Johu from openinverter forum uses a beaglebone successfully for real world charging. I use Raspberry for the light bulb demo charging. So the short anser is: yes.

Fatih738, 2023-09-20:
I have changed pevmac with my mac adress in pyPlcHomeplug.py
run with "sudo python3 pyPlc.py" command
I m getting this error
what is the local mac
Also what is the cause of this error
thanks a lot for your supports
[image]

Fatih738, 2023-09-21:
I have connected raspberry pi to qca7000(red-carrier-board-p) with spi lines. I didnt use the ethernet cable. I will not use because of cp pe pins in the real world charge process. right?

uhi22, 2023-09-21:
Yes, you do not need the ethernet cable.

closed 2023-11-28 by Fatih738