SmartEVSE/SmartEVSE-3 #50: Lock/Unlock cable by MQTT or HTTP request

jongerenchaos, 2024-01-17:
It seems that currently, it is possible to lock and unlock the cable using the RFID tag, as well as through initiating and stopping the charging process in the source code. Would it be worth considering adding a feature where the cable automatically locks upon connection and can be subsequently unlocked through an MQTT or HTTP request?

dingo35, 2024-05-31:
Your use case is not very clear (why would you want this?), but we made the actuator more user friendly in coming 3.6.1: if the car is connected, it locks; if the car is disconnected, it unlocks. 

Does that answer your use case?

dingo35, 2024-06-20:
Closed due to enthousiastic reactions of actuator users to new behaviour, and inactivity.

closed 2024-06-20 by dingo35

jongerenchaos, 2024-06-20:
In the current configuration of SmartEVSE, the cable does not remain locked when the car is connected; it only locks during the charging process. I have modified the code so that the cable locks on the SmartEVSE side immediately when the car is connected, regardless of the SmartEVSE's program status. This means that the cable remains locked at all times and is only unlocked when the car releases the cable and the communication with the car is terminated. At that point, the SmartEVSE also releases the cable. Is this the current configuration of the new version?

dingo35, 2024-06-20:
Yes it is from v3.6.1 on.

jongerenchaos, 2024-06-20:
Tnx i will try that!