SmartEVSE/SmartEVSE-3 #52: Implementing a clock and schedule for minimum morning charge when in Solar mode.

scspijker, 2024-03-25:
### Description
Context: My car has a very large range (400km) and I only need about 80% (60kWh) per week to remain full.

I like the solar mode on SmartEVSE, only charging solar, I've though of a further improvement though.

Currently I just leave it in Solar mode all the time, and plug in everytime I park the car. This usually fills up the car (limited to 80%) to more than I need. Sometimes however, we have little sun for a lot of days, and my charge creeps down to 15%. Then I need to manually tell the charger to charge from grid, however, usually I think of this too late. I'd like to wake up in the morning to a minimum charge of 40%.

I think implementing a clock, and a rudimentary schedule would solve this issue. For example:

Set to solar mode
Set minimum morning charge level to xx%
Set minimum morning charge start time: 02:00 (after sundown, guaranteed in my place of residence).

Say the sun doesn't provide enough power to get my car up and over 40%, the SmartEVSE will start charging at 02:00 from grid power up to xx%, and then stop (unless solar power is available again).

This will make solar mode much more attractive during periods of low sun.

### TL;dr

SmartEVSE in solar only mode.
Sun does not shine enough, car not charged above 40%.
SmartEVSE will start charging using grid power at night until 40% has been reached.

mstegen, 2024-03-25:
There is a thread about adding a modem to the SmartEVSE v3 here:
https://github.com/SmartEVSE/SmartEVSE-3/issues/25
With the modem it's possible to read the SOC of the car.

We are also planning on adding OCPP to the SmartEVSE once firmware version 3.5 is stable.

dingo35, 2024-05-31:
As stated before, for this we need the SoC, which is not available in SmartEVSE-v3 ; if you have that information available in external software (e.g. via a HomeAssistant integration for your specific EV) you can automate this yourself in that software.

closed 2024-05-31 by dingo35