SmartEVSE/SmartEVSE-3 #55: Device not accessible

12koen, 2024-06-12:
**Describe the bug**
Device rarely accessible through web interface running firmware version 3.6.1. Last week it would take a day or so until device was reachable again but last days haven't been able to connect to it at all. Using HA addon but no connection there either. Device is pingable. Charging works as expected.

**Upload your config**
Cannot load config - will update if I can access the device again

**Upload a debug log**
See above

**To Reproduce**
Steps to reproduce the behavior:
1. Device running in Smart mode with sensorbox 2, issue was there before attachment of sensorbox.

**Expected behavior**
I have no way of reaching or resetting the device

**Screenshots**

dingo35, 2024-06-12:
Pleasr upgrade to 3.6.2, its webserver is improved

12koen, 2024-06-13:
Where to find it? Latest released version still 3.6.1

12koen, 2024-06-17:
Seems more stable at 3.6.2, thanks for the update!

mstegen, 2024-06-19:
Closing this now, thanks for your feedback.

closed 2024-06-19 by mstegen