SmartEVSE/SmartEVSE-3 #57: Current limit to 16A?

danny-la, 2024-06-19:
I can charge up to 20A, but since the last update it is limited to 16A. Has there been a limitation somewhere by accident?

dingo35, 2024-06-19:
Not that we are aware of; can you post your /settings here, and if possible, your verbose telnet log?

danny-la, 2024-06-19:
[image]

danny-la, 2024-06-19:
[image]

danny-la, 2024-06-19:
{"version":"v3.6.2","serialnr":5018,"mode":"NORMAL","mode_id":1,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"Deco-wifi","rssi":-75,"bssid":"B0:A7:B9:5A:63:76"},"evse":{"temp":26,"temp_max":65,"connected":true,"access":true,"mode":0,"loadbl":0,"pwm":272,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Ready to read card","rfid_lastread":"000000000000"},"settings":{"charge_current":160,"override_current":0,"current_min":6,"current_max":20,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":600,"solar_max_import":0,"solar_start_current":2,"solar_stop_time":3,"enable_C2":"Not present","modem":"Not present","mains_meter":"Sensorbox","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"","port":1883,"topic_prefix":"SmartEVSE-5018","username":"","password_set":false,"status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":445,"L1":148,"L2":144,"L3":153,"last_data_update":1718830644,"original_data":{"TOTAL":445,"L1":148,"L2":144,"L3":153}},"backlight":{"timer":120,"status":"ON"}}

mstegen, 2024-06-19:
The max charge current is limited by the Circuit menu setting.
It's now set to 16A, if you increase that to 20A (and your breaker + wiring allows it!) The car should charge at 20A.

In Normal Mode, and no power sharing this option is not visible on the Menu. So you'll have to switch to Smart for a moment to set it.
But in Normal mode without power sharing, this should not be necessary. The Max setting should be the limiting factor.
I think it's a bug. Do you agree @dingo35 ?

dingo35, 2024-06-20:
Agreed, in Normal mode, with Pwr Share, Max Circuit should be ignored.

dingo35, 2024-06-20:
Fixed in 8f0b108da in dingo35 repo, will appear soon in this theatre!

closed 2024-06-20 by dingo35