SmartEVSE/SmartEVSE-3 #58: Also wrong current limit on Smart

danny-la, 2024-06-26:
Please check bug #57

I have loaded the new version on Dingo35. For normal charging, I have now the 20A. If I switch over to smart I still have only 16A.

dingo35, 2024-06-26:
In Smart mode MaxCircuit can be configured and is supposed to be obeyed.
Look in the manual, it is called "subpanel configuration", more popular "garage configuration" .

closed 2024-06-26 by dingo35

danny-la, 2024-06-26:
Please check this picture. I even forced the current to 20A, but stays at 16A

[image]

dingo35, 2024-06-26:
Can you upload your /settings again?

danny-la, 2024-06-26:
{"version":"v3.6.3","serialnr":5018,"mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"Deco-wifi","rssi":-76,"bssid":"B0:A7:B9:5A:63:76"},"evse":{"temp":44,"temp_max":65,"connected":true,"access":true,"mode":1,"loadbl":0,"pwm":272,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Ready to read card","rfid_lastread":"000000000000"},"settings":{"charge_current":160,"override_current":200,"current_min":6,"current_max":20,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":600,"solar_max_import":0,"solar_start_current":2,"solar_stop_time":3,"enable_C2":"Not present","mains_meter":"Sensorbox","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"","port":1883,"topic_prefix":"SmartEVSE-5018","username":"","password_set":false,"status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":59,"L1":147,"L2":-92,"L3":4,"last_data_update":1719404086,"original_data":{"TOTAL":59,"L1":147,"L2":-92,"L3":4}},"backlight":{"timer":120,"status":"ON"}}

danny-la, 2024-06-26:
I am charging on 1F at the moment...

dingo35, 2024-06-26:
Set MaxCircuit to 20A and your problem will be solved.

The current override will ALWAYS obey the limitations of your settings, for obvious safety reasons.

mstegen, 2024-06-26:
Are you using a fixed charging cable, or a charging cable+ Charging socket?
As the charge current can also be limited by the use of a 16A charging cable.
Somehow this (CONFIG) setting is not in the /settings.

dingo35, 2024-06-26:
You must configure it at the LCD screen of your SmartEVSE.

dingo35, 2024-06-26:
@mstegen I dont think it is the cable because OP has charged 20A before, as he states in #57

danny-la, 2024-06-26:
So, if I have charged 20A before... the settings are 20A...otherwise this could not be possible in my opion. It always has worked on 20A. 

If I switch to normal mode. I have the 20A.. see picture.

[image]

mstegen, 2024-06-26:
I agree with @dingo35, in Smart mode the setting "CIRCUIT" is checked as well. If you change that to 20A, it should work. 
(Just verified, it does work for me)
This setting can not be changed from the web interface, you will have to configure it on the SmartEVSE itself.

danny-la, 2024-06-26:
I cannot acces the EVSE at the moment. (not visible) But I am sure the setup is 20A. Because that was important setting before I had the web interface.

dingo35, 2024-06-26:
Dont you see that settings are reporting that your MaxCircuit is at 16A?

Why do you keep asking for help and then don't follow our advice? Stop wasting our time....

danny-la, 2024-06-26:
Sorry for wasting your time

danny-la, 2024-06-26:
[image]

mstegen, 2024-06-26:
One menu option to the right... The option is called "CIRCUIT"

danny-la, 2024-06-26:
Thank you. That option was only available when I changed Load Balacing to mode Master. Before that I could not see the option Circuit. It was on disabled because I have only 1 charging point. Was that a strange way of thinking?

mstegen, 2024-06-26:
Hmm strange. For me the menu option "Circuit" is available In Smart mode, and in Normal mode when Power Share (Load balancing) is set to Master. That's how it's supposed to work. The CIRCUIT menu option is only available when it makes sense to use it.
Anyway, i'm glad you got it to work!

dingo35, 2024-06-26:
Apologies from my side are in order here, you actually did find a bug in the user interface. Also I am under the impression you did not find the manual, that might have cleared up a lot:

CIRCUIT	(only appears when PWR SHARE set to <Master>, or when PWR SHARE set to <Disabled>
        and Mode is Smart or Solar and EV METER not set to <Disabled>):
        Set the max current the EVSE circuit can handle (load balancing): 10-200A
        (see also subpanel wiring)

So Circuit is not supposed to limit your charge current when pwr share = disabled, and ev meter is disabled... I will fix this.

danny-la, 2024-06-26:
I have used this manual from Stegen
https://www.stegen.com/nl/index.php?controller=attachment&id_attachment=5

Now it is clear. Can I ask something Else? The minimum current is set to 6A but my car can start at 5A. It would help me in case of Solar option to start earlier. Should a open a request for this?

mstegen, 2024-06-26:
The standard does not allow a charging current lower then 6A. But if your car supports charging at 5A, just set the START current (in solar mode) to -5A. And it will start charging (At 6A)
I have noticed that when i set the charge current to 6A, it actually draws 5A