SmartEVSE/SmartEVSE-3 #60: Wrong version in /update page?

ftamminga, 2024-08-15:
The main page displays version v3.6.7, however, the /update page indicates that the current version is 3.6.2.

I installed 3.6.7 twice with the same result. Ctrl-F5 did not help either.

[image]
[image]

dingo35, 2024-08-15:
This is probably a caching problem in your browser.
Close your webpage, open the smartevse page, and then press "Update" button; does the problem still persist?

ftamminga, 2024-08-15:
Ah, I understand what is going on. The displayed version is set as the session variable 'version' in the home page. If you the press the update button, the update page uses this session variable to display the version.
Anyone who would directly browse to the update page, will either get no version at all, or the version that happens to be the value at that moment. Personally, I think this is not the best way to retrieve the version data, but since I now know how it works, I can live with it.

dingo35, 2024-08-15:
I agree its not the best way, but to fire an extra query to the SmartEVSE to prevent this mismatch seemed a bit "expensive" to me ...

closed 2024-09-07 by dingo35