SmartEVSE/SmartEVSE-3 #61: Unable to compile with latest codebase due to undefined references to mongoose.c

arpiecodes, 2024-09-04:
Hello! I see a lot happened in the past months.

Attempting to clone the newest codebase and attempting to build the firmware with `pio run` I'm seeing the following error;

```
Archiving .pio/build/release/libFrameworkArduino.a
Indexing .pio/build/release/libFrameworkArduino.a
Linking .pio/build/release/firmware.elf
/Users/jorrit/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/release/src/mongoose.c.o:(.literal.packed_open+0x0): undefined reference to `mg_unpack'
/Users/jorrit/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/release/src/mongoose.c.o:(.literal.packed_stat+0x0): undefined reference to `mg_unlist'
/Users/jorrit/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/release/src/mongoose.c.o: in function `packed_open':
/Users/jorrit/Documents/GitHub/SmartEVSE-3/SmartEVSE-3/src/mongoose.c:1893: undefined reference to `mg_unpack'
/Users/jorrit/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/release/src/mongoose.c.o: in function `packed_stat':
/Users/jorrit/Documents/GitHub/SmartEVSE-3/SmartEVSE-3/src/mongoose.c:1863: undefined reference to `mg_unpack'
/Users/jorrit/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/jorrit/Documents/GitHub/SmartEVSE-3/SmartEVSE-3/src/mongoose.c:1865: undefined reference to `mg_unlist'
/Users/jorrit/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/release/src/mongoose.c.o: in function `mg_unpacked':
/Users/jorrit/Documents/GitHub/SmartEVSE-3/SmartEVSE-3/src/mongoose.c:1850: undefined reference to `mg_unpack'
/Users/jorrit/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/release/src/mongoose.c.o: in function `packed_list':
/Users/jorrit/Documents/GitHub/SmartEVSE-3/SmartEVSE-3/src/mongoose.c:1877: undefined reference to `mg_unlist'
collect2: error: ld returned 1 exit status
*** [.pio/build/release/firmware.elf] Error 1
```

Has anyone encountered this before? Cannot seem to figure out what's up.

Compiling on a Mac M2 within VSCode.

dingo35, 2024-09-04:
https://github.com/dingo35/SmartEVSE-3.5/issues/108

closed 2024-09-04 by dingo35

arpiecodes, 2024-09-09:
If anyone encounters the same issue in the future, the following solved it for me;

`sudo ln -s /Library/Developer/CommandLineTools/usr/bin/python3 /Library/Developer/CommandLineTools/usr/bin/python`

zulufoxtrot, 2024-11-24:
I encountered the same issue.

In my case I was using pyenv and fixed it by setting a pyenv global version:

`pyenv global 3.11.x` (run `pyenv` first to list available versions)

Then the build suceeded 👍