SmartEVSE/SmartEVSE-3 #62: EV Meter current & power reading not working

dries007, 2024-09-11:
**Describe the bug**

My EVSE is connected via modbus to 3 phase Eastron meter. It can correctly read the energy (kWh) total (and thus also the current session), but it fails to read the current and power.

The same on the on screen display or via MQTT.

This used to work in a previous version. I think I was on 3.6.6 before, but I'm not sure.

**Upload your config**

```json
{"version":"v3.6.10","serialnr":3551,"mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"DriesFi","rssi":-10,"bssid":"60:A4:B7:1E:07:AB"},"evse":{"temp":41,"temp_max":65,"connected":true,"access":true,"mode":1,"loadbl":0,"pwm":102,"solar_stop_timer":0,"state":"Connected to EV","state_id":1,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":60,"override_current":0,"current_min":6,"current_max":32,"current_main":20,"current_max_circuit":25,"current_max_sum_mains":10,"max_sum_mains_time":5,"solar_max_import":6,"solar_start_current":2,"solar_stop_time":10,"enable_C2":"Always Off","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"192.168.0.4","port":1883,"topic_prefix":"SmartEVSE-3551","username":"evse","password_set":true,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Eastron3P","address":11,"import_active_power":0,"total_kwh":3190.199951,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":3190.199951,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0,"last_data_update":0,"original_data":{"TOTAL":0,"L1":0,"L2":0,"L3":0}},"backlight":{"timer":117,"status":"ON"}}
```

**Upload a debug log**

<details><summary>Details</summary>
<p>

```text
(V) (fn_mqtt)(C1) 7 RECEIVED 87:8:10 <- SmartEVSE-3551/Set/MainsMeter
(D) (Timer100ms)(C0) ModbusRequest 4: Request Energy Node 0
(V) (ModbusSend8)(C1) Sent packet address: 0x0b, function: 0x04, reg: 0x0048, token:0x0b040048, data: 0x0002.
(V) (ModbusDecode)(C1) Received packet (7 bytes) 0b 04 04 45 47 6b 48
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x0b Function 0x04 Register 0x0048 Response
(D) (Timer100ms)(C0) ModbusRequest 20: Request EVMeter Current Measurement Node 0
(V) (ModbusSend8)(C1) Sent packet address: 0x0b, function: 0x04, reg: 0x0006, token:0x0b040006, data: 0x000c.
(V) (ModbusDecode)(C1) Received packet (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x0b Function 0x04 Register 0x0006 Response
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=8.7 A Imeasured=8.7 A MaxCircuit=25 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C0) Current is limited by MaxSumMains: MaxSumMains=12A, Isum=10.5A, Temp_Phases=3.
(V) (CalcBalancedCurrent)(C0) phaseLastUpdate=1726095910.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=8.8 A, Idifference=0.5, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=8.8 A.
(D) (CalcBalancedCurrent)(C0) Checkpoint b: Resetting SolarStopTimer, MaxSumMainsTimer, IsetBalanced=8.8A, ActiveEVSE=1.
(V) (CalcBalancedCurrent)(C0) [H]Node 0 = 8.8 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=8.8 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -2 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 8.7 A IsetBalanced: 8.8 A, MainsMeter.Timeout=10, EVMeter.Timeout=64.
(I) (printStatus)(C0) L1: 8.7 A L2: 0.8 A L3: 1.0 A Isum: 10.5 A
(V) (fn_mqtt)(C1) 7 RECEIVED 87:8:10 <- SmartEVSE-3551/Set/MainsMeter
(V) (fn_mqtt)(C1) 7 RECEIVED 88:8:10 <- SmartEVSE-3551/Set/MainsMeter
(D) (Timer100ms)(C0) ModbusRequest 4: Request Energy Node 0
(V) (ModbusSend8)(C0) Sent packet address: 0x0b, function: 0x04, reg: 0x0048, token:0x0b040048, data: 0x0002.
(V) (ModbusDecode)(C1) Received packet (7 bytes) 0b 04 04 45 47 6b 4c
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x0b Function 0x04 Register 0x0048 Response
(D) (Timer100ms)(C0) ModbusRequest 20: Request EVMeter Current Measurement Node 0
(V) (ModbusSend8)(C0) Sent packet address: 0x0b, function: 0x04, reg: 0x0006, token:0x0b040006, data: 0x000c.
(V) (ModbusDecode)(C1) Received packet (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x0b Function 0x04 Register 0x0006 Response
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=8.8 A Imeasured=8.8 A MaxCircuit=25 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C0) Current is limited by MaxSumMains: MaxSumMains=12A, Isum=10.6A, Temp_Phases=3.
(V) (CalcBalancedCurrent)(C0) phaseLastUpdate=1726095913.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=8.9 A, Idifference=0.4, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=8.9 A.
(D) (CalcBalancedCurrent)(C0) Checkpoint b: Resetting SolarStopTimer, MaxSumMainsTimer, IsetBalanced=8.9A, ActiveEVSE=1.
(V) (CalcBalancedCurrent)(C0) [H]Node 0 = 8.9 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=8.9 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -2 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 8.8 A IsetBalanced: 8.9 A, MainsMeter.Timeout=10, EVMeter.Timeout=64.
(I) (printStatus)(C0) L1: 8.8 A L2: 0.8 A L3: 1.0 A Isum: 10.6 A
(V) (fn_mqtt)(C1) 7 RECEIVED 88:8:10 <- SmartEVSE-3551/Set/MainsMeter
(D) (Timer100ms)(C0) ModbusRequest 4: Request Energy Node 0
(V) (ModbusSend8)(C0) Sent packet address: 0x0b, function: 0x04, reg: 0x0048, token:0x0b040048, data: 0x0002.
(V) (fn_mqtt)(C1) 7 RECEIVED 89:8:9 <- SmartEVSE-3551/Set/MainsMeter
(V) (ModbusDecode)(C1) Received packet (7 bytes) 0b 04 04 45 47 6b 50
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x0b Function 0x04 Register 0x0048 Response
(D) (Timer100ms)(C0) ModbusRequest 20: Request EVMeter Current Measurement Node 0
(V) (ModbusSend8)(C0) Sent packet address: 0x0b, function: 0x04, reg: 0x0006, token:0x0b040006, data: 0x000c.
(V) (ModbusDecode)(C1) Received packet (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x0b Function 0x04 Register 0x0006 Response
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=8.9 A Imeasured=8.9 A MaxCircuit=25 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
[... 85 lines cut ...]
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=9.2 A.
(D) (CalcBalancedCurrent)(C0) Checkpoint b: Resetting SolarStopTimer, MaxSumMainsTimer, IsetBalanced=9.2A, ActiveEVSE=1.
(V) (CalcBalancedCurrent)(C0) [H]Node 0 = 9.2 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=9.2 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -2 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 9.2 A IsetBalanced: 9.2 A, MainsMeter.Timeout=11, EVMeter.Timeout=64.
(I) (printStatus)(C0) L1: 9.2 A L2: 0.8 A L3: 1.0 A Isum: 11.0 A
(V) (fn_mqtt)(C1) 7 RECEIVED 93:8:10 <- SmartEVSE-3551/Set/MainsMeter
```

</p>
</details> 

**To Reproduce**

I'm hooked up to MQTT and Home Assistant. I set the mode based on the current state-of-charge of my car, my desired state-of-charge, and which energy tariff is in effect. (Charge in Smart up to the setpoint using cheap rates and switch to solar otherwise.)

I noticed the issue now because my HA dashboard said my ev was used 0W but I was using 2.5kW total power, but I suspect the issue has existed for a bit.

**Expected behavior**

The EV current & power are read out correctly, like they where in the past.

**Screenshots**

[image]

dingo35, 2024-09-12:
When you look at the log, the response if the modbus on EVMeter current measurement is all zeroes. This suggests the Eaatron is actually reporting 0A.
This suggests hardware fsilure to me; to rule that out, could you downgrade to e.g. v3.6.2 and upload that debug log while charging?

dries007, 2024-09-12:
In v3.6.2 at least the power reading work, not amps, but I'm not sure that ever worked.

<details><summary>Details</summary>
<p>

[image]

```text
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=11.2 A Imeasured=11.4 A MaxCircuit=25 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=2.
(V) (CalcBalancedCurrent)(C0) phaseLastUpdate=1726144183.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=11.2 A, Idifference=1.9, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 3 Isetbalanced=11.2 A, IsumImport=0.3, Isum=6.3, ImportCurrent=6.
(D) (CalcBalancedCurrent)(C0) Checkpoint b: Resetting SolarStopTimer, IsetBalanced=11.2A, ActiveEVSE=1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=11.2 A.
(V) (CalcBalancedCurrent)(C0) [H]Node 0 = 11.2 A(CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=11.2 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -2 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 11.4 A IsetBalanced: 11.2 A, MainsMeterTimeout=10, EVMeterTimeout=64.
(I) (printStatus)(C0) L1: 11.4 A L2: -6.0 A L3: 0.9 A Isum: 6.3 A
(D) (Timer100ms)(C0) ModbusRequest 4: Request Energy Node 0
(V) (ModbusSend8)(C0) Sent packet address: 0x0b, function: 0x04, reg: 0x0048, token:0x0b040048, data: 0x0002.
(V) (ModbusDecode)(C1) Received packet (7 bytes) 0b 04 04 45 48 74 c9
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x0b Function 0x04 Register 0x0048 Response
(V) (ModbusSend8)(C0) Sent packet address: 0x0b, function: 0x04, reg: 0x0034, token:0x0b040034, data: 0x0002.
(V) (ModbusDecode)(C1) Received packet (7 bytes) 0b 04 04 45 13 a0 00
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x0b Function 0x04 Register 0x0034 Response
(D) (Timer100ms)(C0) ModbusRequest 20: Request EVMeter Current Measurement Node 0
(V) (ModbusSend8)(C0) Sent packet address: 0x0b, function: 0x04, reg: 0x0006, token:0x0b040006, data: 0x000c.
(V) (ModbusDecode)(C1) Received packet (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x0b Function 0x04 Register 0x0006 Response
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=11.2 A Imeasured=11.4 A MaxCircuit=25 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=2.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=11.2 A, Idifference=1.9, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 3 Isetbalanced=11.2 A, IsumImport=0.3, Isum=6.3, ImportCurrent=6.
(D) (CalcBalancedCurrent)(C0) Checkpoint b: Resetting SolarStopTimer, IsetBalanced=11.2A, ActiveEVSE=1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=11.2 A.
(V) (CalcBalancedCurrent)(C0) [H]Node 0 = 11.2 A(CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=11.2 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -2 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 11.4 A IsetBalanced: 11.2 A, MainsMeterTimeout=8, EVMeterTimeout=64.
(I) (printStatus)(C0) L1: 11.4 A L2: -6.0 A L3: 0.9 A Isum: 6.3 A
(V) (fn_mqtt)(C1) 3 RECEIVED 114:-60:9 <- SmartEVSE-3551/Set/MainsMeter(fn_mqtt)(C1) 3 RECEIVED 114:-60:9 <- SmartEVSE-3551/Set/MainsMeter(Timer100ms)(C0) ModbusRequest 4: Request Energy Node 0
(V) (ModbusSend8)(C0) Sent packet address: 0x0b, function: 0x04, reg: 0x0048, token:0x0b040048, data: 0x0002.
(V) (ModbusDecode)(C1) Received packet (7 bytes) 0b 04 04 45 48 74 cd
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x0b Function 0x04 Register 0x0048 Response
(V) (ModbusSend8)(C0) Sent packet address: 0x0b, function: 0x04, reg: 0x0034, token:0x0b040034, data: 0x0002.
(V) (ModbusDecode)(C1) Received packet (7 bytes) 0b 04 04 45 13 a0 00
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x0b Function 0x04 Register 0x0034 Response
(D) (Timer100ms)(C0) ModbusRequest 20: Request EVMeter Current Measurement Node 0
(V) (ModbusSend8)(C0) Sent packet address: 0x0b, function: 0x04, reg: 0x0006, token:0x0b040006, data: 0x000c.
(V) (ModbusDecode)(C1) Received packet (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x0b Function 0x04 Register 0x0006 Response
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=11.2 A Imeasured=11.4 A MaxCircuit=25 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=2.
(V) (CalcBalancedCurrent)(C0) phaseLastUpdate=1726144187.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=11.2 A, Idifference=1.9, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 3 Isetbalanced=11.2 A, IsumImport=0.3, Isum=6.3, ImportCurrent=6.
(D) (CalcBalancedCurrent)(C0) Checkpoint b: Resetting SolarStopTimer, IsetBalanced=11.2A, ActiveEVSE=1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=11.2 A.
(V) (CalcBalancedCurrent)(C0) [H]Node 0 = 11.2 A(CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=11.2 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -2 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 11.4 A IsetBalanced: 11.2 A, MainsMeterTimeout=10, EVMeterTimeout=64.
(I) (printStatus)(C0) L1: 11.4 A L2: -6.0 A L3: 0.9 A Isum: 6.3 A
(D) (Timer100ms)(C0) ModbusRequest 4: Request Energy Node 0
(V) (ModbusSend8)(C0) Sent packet address: 0x0b, function: 0x04, reg: 0x0048, token:0x0b040048, data: 0x0002.
(V) (ModbusDecode)(C1) Received packet (7 bytes) 0b 04 04 45 48 74 d5
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x0b Function 0x04 Register 0x0048 Response
[... cut ...]
(D) (CalcBalancedCurrent)(C0) Checkpoint b: Resetting SolarStopTimer, IsetBalanced=11.2A, ActiveEVSE=1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=11.2 A.
(V) (CalcBalancedCurrent)(C0) [H]Node 0 = 11.2 A(CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=11.2 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -2 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 11.4 A IsetBalanced: 11.2 A, MainsMeterTimeout=10, EVMeterTimeout=64.
(I) (printStatus)(C0) L1: 11.4 A L2: -6.0 A L3: 0.9 A Isum: 6.3 A
```

</p>
</details>

dries007, 2024-09-12:
Aah that explains some things.

I assume that's why I see a big spike (up to the max main fuse rating) initially, the control algo is not happy. 
My P1 data has more latency than most I assume: I have to send the P1 data via a slow wireless serial link to my home assistant, which then uses an automation to publish the data to MQTT.

Aside from this spike, which only lasts 2~4 seconds, I've had no issues before. I get why it's not the preferred situation, but can the reading of the kW be restored without impact for other users? If not a separate meter type maybe?

I would really like to avoid having to replace the meter. The setup was paid for by my employer (by ordering the full setup from someone). Replacing the meter would mean not only an invoice which is annoying to explain, but also resetting the kWh total, which is used for power reimbursement.

dingo35, 2024-09-12:
You might get the desired result by defining a custom meter with the values that are defined for Eastron3P; the custom meter uses the old routine...

dingo35, 2024-09-12:
Note that a full fledged Eastron SDM72D-M v2 costs €55 incl shipping. The time we spent here is already more ...

dries007, 2024-09-12:
Well, that's the one I already have. The SDM630 seems to support A & V measurements, and that one costs ~€130 from reputable places.

But I'll try that custom meter idea.

dingo35, 2024-09-12:
No you should notice the _v2_ . The v2 supports all params including currents and power per phase.

Not loving the counter-intuitive naming either, but the v2's are used widely with the SmartEVSE.

dries007, 2024-09-12:
Oooh that's great :smile:. I'm sure I'll find a use for the existing meter then.

dingo35, 2024-09-12:
https://www.amazon.nl/V2-energiemeter-lcd-display-achtergrondverlichting-gekalibreerd/dp/B08X96T1X2/ref=mp_s_a_1_1?crid=1VIWARA7T05ZZ&dib=eyJ2IjoiMSJ9.p9tUs6S1zefo74pcNdZdbKc-JUwgRaQPB2G7NEVGxuS0bjJgEhz2BkRcaTtCRoPvdR8JY3XX8MVSIaikPIH3TE9x1uzlaZ1m2GtiyEC5l9XE9icmytDPDsA1KTI-NA1Y4y8vT3kQt1JOu0ApJpYQq_3-dRcj948Pzn6xll55frc.rLDnpZGAmFNYRNr45O_r1qFGaHKdiw6T2is9pO8_wxU&dib_tag=se&keywords=eastron+sdm72d-m+v2&qid=1726164156&sprefix=eastron+%2Caps%2C345&sr=8-1