SmartEVSE/SmartEVSE-3 #64: Curl and MQTT commands don't act as supposed

HeinvdW, 2024-10-02:
**Describe the bug**
A clear and concise description of what the bug is.
Some Curl commands are unclear.
Some MQTT commands are unclear

**Upload your config**
Press the "raw" button in the SmartEVSE webserver and upload the json settings file, so we can look at your configuration.
{"version":"v3.6.10","serialnr":75xx,"mode":"OFF","mode_id":0,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"ZIGGO_3AE24","rssi":-73,"bssid":"BE:30:D9:40:D1:B5"},"evse":{"temp":16,"temp_max":65,"connected":true,"access":false,"mode":0,"loadbl":0,"pwm":1024,"solar_stop_timer":0,"state":"Charging Stopped","state_id":9,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":60,"override_current":0,"current_min":6,"current_max":16,"current_main":16,"current_max_circuit":16,"current_max_sum_mains":0,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":4,"solar_stop_time":10,"enable_C2":"Always Off","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"192.168.2.xx","port":1883,"topic_prefix":"SmartEVSE","username":"","password_set":false,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0,"last_data_update":0,"original_data":{"TOTAL":0,"L1":0,"L2":0,"L3":0}},"backlight":{"timer":0,"status":"OFF"}}

**Upload a debug log**
Flash the debug version of the firmware (see https://github.com/dingo35/SmartEVSE-3.5/blob/master/SmartEVSE-3/HowToFlash.txt), telnet to your device, capture the debug log, cut it so that it shows JUST BEFORE and JUST AFTER the problem arises, and upload it here.

**To Reproduce**
Steps to reproduce the behavior:
1. What Mode are you in? Normal, Smart or Solar?
2. What steps do you take so the problem arises?
The Curl command: 
curl -X POST http://192.168.2.77/settings?mode=0 -d '' works values 0,1 used
The MQTT command
'mosquitto_pub  -h 192.168.2.109 -t "SmartEVSE/Set/Mode" -m 0' values 0 and 1 used
doesn't give a reaction
However the MQTT command
'mosquitto_pub  -h 192.168.2.104 -t "SmartEVSE/Set/CurrentOverride" -m 60' works as expected
and curl -X POST http://192.168.2.71/settings?CurrentOverride=60 -d ''
nor curl -X POST http://192.168.2.71/settings?ChargeCurrentOverride=60 -d ''
show any reaction in the MQTT dump

**Expected behavior**
A clear and concise description of what you expected to happen; expected Charging Current?
I expect both Curl and MQTT to react.
Curl does not accept some commands and MQTT does not either.

It would be nice if the docs https://github.com/SmartEVSE/SmartEVSE-3/blob/master/docs/configuration.md, the REST API and the MQTT API could be updated with the proper commands that are accepted by the device.

Regards

PS I'm using Normal mode and send commands from Domoticz.

dingo35, 2024-10-02:
Thanks for your report; this is a typical thing that does not need any development knowledge; perfect for a community effort! You wouldnt want to leech only, would you?

I am looking forward to your Pull Request!

HeinvdW, 2024-10-02:
> [Tracking afbeelding][https://github.com/notifications/beacon/AGA2NBKDFE62CYP7OZRAJELZZR5KLA5CNFSM6AAAAABPIRTV5CWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUOOP6LQ.gif]

Hi,
Thanks for your answer.
I'm quite new on GitHub and honestly I don't understand what you intend to say.
Leech means take profit.
Ok,
But what should I do?
There's something missing in the docs.
I don't know how to add the right text as I don't know the proper commands.

What is a pull request supposed to do.

Please explain. I feel a bit stupid not understanding how.

Regards

dingo35, 2024-10-03:
You are so lucky to live in the ages of the internet; no need to bike to a library, Google is your friend...

Some netiquette: you only ask questions you have searched for, and tried out yourself. So it is more about reading, then writing...

Keywords to look for: github fork repository, git clone repository, git commit changes, git push changes to repository, github create pull request.

There are tons of blogs and examples out there, and thousands, if not millions, went before you and educated themselves; we are standing on the shouldere of giants, you just have to climb up...

Good luck on your path to knowledge!

closed 2024-10-03 by dingo35