SmartEVSE/SmartEVSE-3 #66: access to EVSE over USB-C port

Benthouse, 2024-10-26:
All,

got a new EVSE yesterday after the previous one was broken due to too high voltage on the ModBus connections.
This morning I tried to put the settings in. As I have an RFID card reader on the EVSE I tried to learn my card without success. Got the message that the RFID memory was already full?? As WiFi wasn't configured yet I don't have access now to the device as the device is asking for a RFID card/tag which is stored in the memory which I don't have.
Is there an easy way to get access to the device over the USB-C connection as WiFi isn't yet activated? Need to go back to standard settings to get access again. Thougt this could be done with a Firmware flash.

Can somebody explain in easy steps how to solve my issue as currntly I can't do anything yet. I took the device out of the cabinet and now it is giving the message 'Residual Fault Current Detected' as the RCMON is not connected.

Is there an easy workaround??

This should be normal in my point of view in the system however I can't get acces; even not eith the keys on the front of the equipment.

thanks for your support

Benthouse, 2024-10-26:
> All,
>
> got a new EVSE yesterday after the previous one was broken due to too high voltage on the ModBus connections. This morning I tried to put the settings in. As I have an RFID card reader on the [...]

issue solved with connecting a piece of wire in between 0V and SW (In1).

dingo35, 2024-10-26:
Ok, glad you solved it. For any other user reading this, there is always a backup procedure, if you connect your device through USB and you give <enter>, it will ask you for wifi SSID and password, so you can ALWAYS get connected and proceed...

closed 2024-10-26 by dingo35