SmartEVSE/SmartEVSE-3 #67: Solar mode power regulation

TFleury, 2024-11-01:
**Describe the bug**
When using solar mode with `solar_max_import` parameter greater than 0, the EVSE tries to reach `solar_max_import` value instead of 0.

For example, if you set "current_min": 6, "solar_start_current": 2 and "solar_max_import": 5
When mains meter drops below -2A charging starts, then it regulates charging power to keep mains meter around +5A.

**Expected behavior**
Charging current should stay at `current_min` (6A in my exemple) as long as mains meter don't drop under 0A.

dingo35, 2024-11-01:
Well your expectations are wrong: solar_max_import is exactly that.

I recommend setting solar_max_import to 0A, the solar timer to 60 minutes, and startcurrent at your preferres value, and you will see the behaviour you want!

TFleury, 2024-11-01:
Sorry, but the documentation says :
> solar_max_import
> 
>   The maximum current (sum of all phases) of the MainsMeter that can be imported before the solar timer is fired off,
>   after expiration the car will stop charging.

So, it's a maximum not a target value and it's supposed to only trigger stop timer, not to regulate at that value.
Maybe it needs an additional parameter like `solar_import`

bobosch, 2024-11-03:
@TFleury I also prefer your version of this option, and i added a compile flag to  the Smartevse V2 version some years ago
https://github.com/SmartEVSE/SmartEVSE-2/commit/b784585f1a3091bb39330b42a2c2f88c417d3b3e

bobosch, 2024-11-03:
Yes, there should be two options.
- The balance point (positive for steady import, negative for steady export)
- The solar import (to allow start charging before reaching the 6 A surplus)

I also added a compile flag to change the solar import to a per phase value on the V2 version
https://github.com/SmartEVSE/SmartEVSE-2/commit/a58c5cc8a43e6416b9944f0efc9aade767d5cdab
because when configuring an import of e.g. 15 A, a 3 phase car waits until some solar power is available, while a 1 phase car starts charging at full speed...

dingo35, 2024-11-03:
Interesting; but you seem to prefer ImportCurrent = 0 (looking at #ifdef SPECIAL), so then #ifdef ALWAYS is not making any difference anymore ?!

TFleury, 2024-11-04:
@bobosch thank you for your feedback.
I think the two parameters approach is more interesting than compile flag which involves maintaining a different build.
Are you interested in a PR for that ?

bobosch, 2024-11-06:
> but you seem to prefer ImportCurrent = 0 (looking at #ifdef SPECIAL), so then #ifdef ALWAYS is not making any difference anymore ?!
This was existing code, I don't touched it ...

bobosch, 2024-11-07:
Maybe It's enough to change the behavior of the START option

{"START",   "Surplus energy start Current (per charging phase)", 0, 16, START_CURRENT}
When set to e.g. 1 A it should also allow to draw 5 A from grid (and initiate STOP timer when drawing more)
as mentioned before it must be per charging phase for same behavior...

I have only V2 to test - I will implement it there first. When it works I will need some help to check it on V3

TFleury, 2024-11-07:
Good idea, it's simple and don't requires new setting.
I will be able to implement and test it on V3.

bobosch, 2024-11-17:
New commits on V2
[c8f65863ddb3d4aa9a8f2f00ed88df26d871c035](https://github.com/SmartEVSE/SmartEVSE-2/commit/c8f65863ddb3d4aa9a8f2f00ed88df26d871c035) Allow ImportCurrent to be negative (for steady export)
[544a635e7296521e88df59f2ad4e5081dfeb302f](https://github.com/SmartEVSE/SmartEVSE-2/commit/544a635e7296521e88df59f2ad4e5081dfeb302f) Allow StartCurrent to be negative
[ba6415511a215285029a056febed5d8d93db95e0](https://github.com/SmartEVSE/SmartEVSE-2/commit/ba6415511a215285029a056febed5d8d93db95e0) Make StartCurrent per phase (for same behaviour on different cars)
[bc95b8d1f2d2619e3501afd7e71c55575666c4cc](https://github.com/SmartEVSE/SmartEVSE-2/commit/bc95b8d1f2d2619e3501afd7e71c55575666c4cc) Stop solar charging only when over 6 A (per phase) more than StartCurrent is used

bobosch, 2024-11-17:
I tried to merge these commits into V3:
[tmp.zip](https://github.com/user-attachments/files/17790163/tmp.zip)

TFleury, 2024-12-02:
I think the negative value handling is a bit tricky and it will introduce breaking changes with settings storage and web API.
I just proposed a PR (#70) to avoid start/stop cycles.

dingo35, 2025-04-01:
Handled in #70

closed 2025-04-01 by dingo35