SmartEVSE/SmartEVSE-3 #68: Master branch is broken

TFleury, 2024-11-03:
Hi all,

I just started to play with my Smart EVSE and I'm surprised that compiling and uploading current main branch, my Smart EVSE is no more functional : "No Power Available" error.
It seems that there is a V4 version on the way, which seems to be based on slightly different hardware, so could it be possible to develop it on separate branch or repository while in progress and keep main branch functional ?

dingo35, 2024-11-03:
Thanks for your report Thierry!

If you compile with "pio run - t upload" you should have the exact v3 version you always had; the v4 development should have no effect, BUT the introduction of the OCPP functions might.... 
So if you compiled it like that, and you still get that error, you found a bug. If you didn't, what compile instructions did you use?

I cannot reproduce that bug; could you please post your /settings and the telnet log, starting approx. 12 seconds before the timeout happens, until after the moment it has been done?

TFleury, 2024-11-03:
Thank you for you answer.

I found the issue with debug log : for some unknown reason the `current_max_sum_mains` was set to 10, setting it back to 0 solved the issue.

To go back to my question, is the main branch assumed to be stable ?

dingo35, 2024-11-03:
No, the main branch is in development. Even the release on dingo35 repo is assumed "bleeding edge"; the release on smartevse repo is "stable".

Browse to /update of your smartevse for clarification.

closed 2024-11-04 by TFleury