SmartEVSE/SmartEVSE-3 #69: Add time blocks for max consumption during the day

stricklandcleveland, 2024-11-04:
More and more countries are implementing time blocks over the day where you have a limit of max electricity consumption. If you go over the limit you pay extra fee.

Look at this picture for example:

[image]

The feature request here is that there would be an option to set those limits for blocks in kW. It can be done over web UI as it will be much easier than on the small display. The unit already reads the data of house energy consumption over modbus or MQTT so that part is already covered. 

With this feature we would be changing CurrentMaxSumMains based on the scheduled time blocks with max kW defined.

devdems, 2024-11-04:
Great suggestion. We have those exact time blocks and setting this directly on SmartEVSE would be ideal because not a lot of people have HA.

dingo35, 2024-11-04:
So please be ultimately clear: if the energy consumption is to be limited, we are talking about a max of kWh in a certain time block.

If the power consumption is to be limited, we are talking about a max of kW _during any time in a cerain time block_ .

Which is it?

stricklandcleveland, 2024-11-04:
You have 5 time blocks.

Limits in time blocks (TB) are:
TB1: 7kW
TB2: 9kW
TB3: 10kW
TB4: 10kW
TB5: 13kW

What that means is that you cannot exceed the max limit power set for that TB in a average demand during 15min blocks inside that TB.

For example: Currently it's TB2.
Average demand in 15min cannot exceed 9kW.

I don't think we want to store power for 15min to calculate average demand so to make it more simple to implement i would leave those 15min block average aside and just focus at the current usage for the TB. 

Therefore based on the house usage we would just limit the EV charger so that current total usage does not exceed the limit set for current TB. If we do that then it's not possible that the average demand in 15 min will exceed the limit.

odem881, 2024-11-05:
This would be a very good addon. I would also recommend that we add an option to update schedule/limits over MQTT.

dingo35, 2025-04-01:
From a technical point of view:
"I don't think we want to store power for 15min to calculate average demand so to make it more simple to implement i would leave those 15min block average aside and just focus at the current usage for the TB."

Power x Time = Energy; or in simpler terms: kW x h = kWh .

So we don't need to calculate average power, if the mainsmeter gives off Energy measurements; we just record the energy measurement at the beginning of the time block, and stop charging when the max. Energy limit is reached. 
Now the problem with that is that the other electricity users in the house would exceed the Energy limit , so we need to put a margin in for the non-regulated house users?

That way you wouldn't even have to configure all that PV+Minimum Power, Exclusive PV or Grid + PV shit; you just guard your mainsmeter values...

Or am I misunderstanding something?

odem881, 2025-06-04:
Do we have any possible timeframe for when this setting of max 15min kWh per time block can be added?