SmartEVSE/SmartEVSE-3 #73: Smart EVSE shows "Out of Order" with Red light on push button.

bjolyon, 2024-12-10:
Fully functional Smart EVSE running in Solar Mode with Modbus inout from smart meter suddenly (when not charging and not connected to car) showed "Out Of Order". Disconnecting and reconnecting to mains continues to show "Out of Order". What circumstances can lead to this message?

bjolyon, 2024-12-10:
MStegen, In the meantime I swapped Smart EVSE with spare unit I had. Identical configuration, running 3.7.2 Replacement worked fine. 
I then put "Out of Order" Smart EVSE on the bench with USB3 power in, no other connection; still shows "Out of Order". Must be hardware fail problem. Please advise if worth returning for repair (I am in Australia, so mailing cost issue!).

mstegen, 2024-12-11:
Can you try to erase the settings on the controller by going to http://ip-address/erasesettings ?
where "ip-address" is the IP address of the controller.
This will bring the controller back to it's default settings.

bjolyon, 2024-12-11:
Thanks, Michael, I performed the erase settings command and it has come back to normal function! Remarkable (to me!).
So many thanks for the solution. Can you speculate what might have occurred to cause the controller to spontaneously go to an "Out of Order" state?

closed 2024-12-12 by mstegen