SmartEVSE/SmartEVSE-3 #75: Ford Kuga has charge issues

evertvanes, 2025-01-22:
I have two vehicles that I regularly connect to my smartevse v3 charger using the latest software. A BMW works very well with smartevse, I have not had any issues. But our Ford Kuga does not like the way the smartevse works.

With some persuation I can charge the Ford Kuga but almost all sessions end up in error. And connecting the Kuga to the cable must be done in a certain order or the charge will not start. If I connect the cable before the smartevse is active the Ford will error and only continue by disconnecting the cable first, then enable smartevse and connect. Only then the charge will start. As soon as I disable the smartevse while it is connected the Ford will error.

In order to diagnose the Ford issues I have a raspberry pi which uses telnet to log all smartevse messages to syslog. If any one needs this functionality just install screen and start a new session. In this session start the command:
telnet 192.168.1.10 | logger -t smartevse
now put the session in the background, you can then look at syslog file while messages are collected.

My setup uses a separate p1 monitor and the API is used to send the L1/2/3 values to the smartevse. And I have a fixed cable outside my house, I use a switch to disable or enable the smartevse

When I look at all the messages, I found that last night at 03:48 there was a State change from B to B1 because the API had a timeout, for 30 seconds there was no P1 information. The state change to B1 also disconnects the contactor, even when the car is not charging. I think the Ford Kuga does not like the power being disconnected.

Any thoughts on what to do to further diagnose and maybe solve the situation?

Anyone with experience on charging Ford Hybrid vehicles?

Any help appreciated. Very happy with the hardware and software!

dingo35, 2025-01-22:
The protocol demands that when switching to B1 or C1, the contactor is switched off; but the EV should be informed by the CP pin that power is going to be switched off....

You are answering your own question: prevent this situation by providing a reliable feed of the mains currents; if you are not able to do that you should charge in Normal mode...

If you want to investigate to improve robustness, you could try flashing the first v1.0.0 version of Stegen, or the v1.8.0 version, see if they behave differently then current v3.7.3; you can find all of them in the dingo35 repo, but I recommend this route only if you are familiar with platformio to build and flash the downgraded firmware.

Hans-Mep, 2025-01-30:
Still pending delivery of my Ford Kuga 2025. Charger based on SmartEVSE (v3.7.3) and EVCC in testing phase with home brew tester. Software (e.g. EVCC, MBMD) running on Raspberry Pi. Interfacing through shell scripts. P1 through sensorbox v2 and RS485. EV meter  based on SDM630 modbus/Waveshare/MBMD and MQTT API feed to SmartEVSE.
Ford Kuga only is 1 phase 16 amp charging. SmartEVSE is giving me an error when I use L2 instead of L1 (EVmeter error). Needs more investigation. Not sure if this can result in your error.
Ford NL officially is not supporting dynamic charging for the Kuga yet it seems. They mention Tibber as potential solution but not sure if that will work with the Kuga (its hybrid and not full electric)
As soon as my Kuga will be available I will start real life testing including FordPass capabilities.

dingo35, 2025-02-04:
Imeasured is of no importance when you are in Normal mode.

If the SmartEVSE would be in malfunction, it would switch to state C1 (from C) or B1 (from B).

The fact that it is switching from C to B shows that the EVSE thinks everything is ok, while the EV thinks there is a malfunction.

That looks like either incorrect connection of the CP line (or PE), or a bug in the Ford firmware...

Hans-Mep, 2025-02-05:
Compared to specs it seems your Ford Kuga initial is pulling some higher charge current then expected (16.1 Amp vs 16 Amp).
In SmartEVSE under "Normal" mode you have to configure "MAX" which, according to description, should be the max charge current SmartEVSE can deliver.
On the other hand, SmartEVSE module is delivering the PWM signal through the CP line if I'm correct. Maximum charge current is limited by the CC line (internal resistor in case of fixed cable and EVSE responsibility in case of connector).  
What if, to test, change the EVSE MAX current setting temporary to 17 Amp?
As I have my mains circuit for the charger fused with 16 Amp, personally I expect issues also when this is correct.
Is your standard SCHUKA charger working correctly?
Hope to start real life testing within 2 weeks. Worst case I will re-start my discussion with Ford-NL. To be clear: Pre-sale I was not very impressed by their reaction on dynamic charging of the Kuga.

dingo35, 2025-02-06:
Try setting MinCurrent to 7A, this helped with a Kona:
https://github.com/dingo35/SmartEVSE-3.5/issues/40#issuecomment-2640681624

mstegen, 2026-02-07:
Hello @evanes68 did you fix the issue with the Kuga?
If so please close the issue. Thanks!