SmartEVSE/SmartEVSE-3 #76: Feature request: Default RFID card for the OCPP authentication

Timple, 2025-02-16:
**Is your feature request related to a problem? Please describe.**
I've got the RFID with OCPP working (yeay, thanks for this feature 🙏 ). However charging with RFID card is twice as much effort as without 🙂 . As I need to plug the vehicle and use the card.

**Describe the solution you'd like**
Set a default RFID card which is always used for the OCPP authentication.

**Describe alternatives you've considered**
Perhaps if incidental other cards should be used (visiting family members) they could authenticate before plugging. Then the default card would not be used. If no card was used before the plugging, the default card is sent to the OCPP.

**Additional context**
I am the sole use of my charging station. So I always need the same card.

dingo35, 2025-02-16:
Have you tried checking the "auto authorization" box ?

dingo35, 2025-02-16:
And/or disabling your RFIDreader?

Timple, 2025-02-16:
Ow, that's briliant! There is such a function then...

I checked the logs from my OCPP provided and copied my tagId from there. But when I enter this exact tagId into the autorization box, it fails.
The display says "Invalid RFID card" "Finished Charging"

When I look in my OCPP provider logs, I see a different hexadecimal number. Which is obviously rejected.

Is there any conversion being done with the default tagId?

dingo35, 2025-02-16:
No not the backend URL, not the chargebox Id, but the AUTO-AUTHORIZE CHECKBOX.

Please read my answers thoroughly!

And combine the checkbox with the settings RFIDreader=Rmt/OCPP and Disabled.

So there are 4 combinations you must try; one of them probably works with your backend provider....

Timple, 2025-02-17:
> No not the backend URL, not the chargebox Id, but the AUTO-AUTHORIZE CHECKBOX.

I did, I only modified that checkbox and the "Default idTag" field that pops up.

> So there are 4 combinations you must try

I assume that would be these four?

| Auto auth  | RFID | |
| - | - | - |
| Off | On | This works, but I have to swipe the card
| Off  | Off | I don't have a field to put my default tag id in
| On  | Off | Untested, will do that tonight
| On  | On | "Invalid RFID card" error

> one of them probably works with your backend provider....

I don't see yet how my backend provider is of influence. I would like the EVSE to present the UUID as if an RFID card was swiped.

---
Thanks for the reply, I will verify the untested option, so I'll get back to this later!

dingo35, 2025-02-17:
The Off/Off version might work for you, as you saw in the code there is a "FreeUse" mode in OCPP where you dont provide an RFID tag, but you can still charge and your OCPP provider still keep track of the charges.

Could you pm me:
- the idTag as reported by your provider when it is accepted at Off/On test
- the last read RFID as reported immediately after the test by http://ipadress/settings of your SmartEVSE
- the idTag as reported by your provider when it is rejected at On/On test
- again the last read RFID

Please do it per pm since you dont want people to emulate your card and start charging at your cost ...

Timple, 2025-02-19:
Update: The untested scenario (auto-auth: on, rfid off) unfortunately also gives an "Invalid RFID card" error.

I'll send @dingo35 the details as requested!

dingo35, 2025-02-20:
I can reproduce the problem, Ill forward it to the dev of the OCPP module...

Timple, 2025-03-20:
Cool, let me know if I need to test anything!

dingo35, 2025-03-30:
Ok turns out I can NOT reproduce this problem; on my test bench the EV is always (simulated to be) connected, so we always got from state B->C and back; turns out starting from state A (which is everyday behaviour) all works fine with Auto-Authorize on, freeVend IdTag defined and RFIDReader=Rmt/OCPP.

The debug output of the OCPP module can only be seen via your USB serial cable; are you willing and able to produce some logs for me that way?

Timple, 2025-03-30:
Sure!

Please provide some steps and I'll go ahead!

dingo35, 2025-04-01:
Ok goto http://smartevse-xxx.lan/update and update to v3.7.4 DEBUG VERSION!!! 

Now connect your SmartEVSE to your laptop through the USB-C interface; run a screen terminal program on your laptop, e.g. on mine (Linux) it is:
screen -L /dev/ttyUSB0 115200

Now when you check the "Enable OCPP" checkbox, you should see on your terminal messages like this:

[image]

 Now log what happens if you enable Auto-authorize and fill in a made up tagID, like 11223344556677; see what the log says and what your backend provider reports, and post it here.

You can do the same experiment with your own tagID, BUT DO NOT POST IT HERE if you use that tag on public chargers; you don't want someone to spoof your tagID and start charging on your bill.....

Timple, 2025-04-01:
"Enable OCPP" checkbox:

```
[MO] Send: [2,"7fd98f49-060a-6147-1d74-0fcac62b0d7c","BootNotification",{"chargePointModel":"SmartEVSE","chargePointSerialNumber":"7767","chargePointVendor":"Stegen Electronics","firmwareVersion":"v3.7.4","meterType":"Eastron3P"}]
[MO] Recv: [3,"7fd98f49-060a-6147-1d74-0fcac62b0d7c",{"status":"Accepted","currentTime":"2025-04-01T17:07:46.651Z","interval":900}]
[MO] Send: [2,"f67b3e72-e527-d990-0919-8127725ff970","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2025-04-01T17:07:46Z"}]
[MO] Recv: [3,"f67b3e72-e527-d990-0919-8127725ff970",{}]
[MO] Send: [2,"9d6c74ad-bded-feb2-e93e-0b2d5c05445d","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2025-04-01T17:07:46Z"}]
[MO] Recv: [3,"9d6c74ad-bded-feb2-e93e-0b2d5c05445d",{}]
```
___

"Auto-authorize" (11223344556677)

No output...

___

Plug in vehicle:

```
[MO] Send: [2,"82cf0a87-1170-4ee3-65ab-bbfeb21e923e","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2025-04-01T17:15:03Z"}]
[MO] Recv: [3,"82cf0a87-1170-4ee3-65ab-bbfeb21e923e",{}]
[MO] Send: [2,"3888b809-f7a9-8128-94f8-60a1083888b9","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"SuspendedEV","timestamp":"2025-04-01T17:15:03Z"}]
[MO] Recv: [3,"3888b809-f7a9-8128-94f8-60a1083888b9",{}]
[MO] Send: [2,"b177a108-9305-31f7-1871-17024ee315b3","StartTransaction",{"connectorId":1,"idTag":"11223344556677","meterStart":560298,"timestamp":"2025-04-01T17:15:03Z"}]
[MO] Recv: [3,"b177a108-9305-31f7-1871-17024ee315b3",{"idTagInfo":{"status":"Invalid"},"transactionId":10038}]
[MO] Send: [2,"25247ce4-ca5a-561b-d6c7-ac34ed9cc4dd","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Charging","timestamp":"2025-04-01T17:15:03Z"}]
[MO] Recv: [3,"25247ce4-ca5a-561b-d6c7-ac34ed9cc4dd",{}]
[MO] Send: [2,"51d74167-29e1-65e3-5b3c-2b0d7b3dcb6d","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"SuspendedEVSE","timestamp":"2025-04-01T17:15:04Z"}]
[MO] Recv: [3,"51d74167-29e1-65e3-5b3c-2b0d7b3dcb6d",{}]
[MO] Send: [2,"9d6c74ac-593e-0b2e-24f8-60a108388960","StopTransaction",{"meterStop":560298,"timestamp":"2025-04-01T17:15:09Z","transactionId":10038,"reason":"DeAuthorized"}]
[MO] Recv: [3,"9d6c74ac-593e-0b2e-24f8-60a108388960",{"idTagInfo":{"status":"Invalid"}}]
[MO] Send: [2,"e315b46d-b9d6-74ad-bded-ff5ab6bbfeb3","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Finishing","timestamp":"2025-04-01T17:15:09Z"}]
[MO] Recv: [3,"e315b46d-b9d6-74ad-bded-ff5ab6bbfeb3",{}]
```

Provider output:
```json
{"chargePointModel":"SmartEVSE","chargePointSerialNumber":"7767","chargePointVendor":"Stegen Electronics","firmwareVersion":"v3.7.4","meterType":"Eastron3P"}
{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2025-04-01T17:07:46Z"}
{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2025-04-01T17:07:46Z"}
{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2025-04-01T17:15:03Z"}
{"connectorId":1,"errorCode":"NoError","status":"SuspendedEV","timestamp":"2025-04-01T17:15:03Z"}
{"connectorId":1,"idTag":"DE8EFB6B","meterStart":560298,"timestamp":"2025-04-01T17:15:03Z"}
{"connectorId":1,"errorCode":"NoError","status":"Charging","timestamp":"2025-04-01T17:15:03Z"}
{"connectorId":1,"errorCode":"NoError","status":"SuspendedEVSE","timestamp":"2025-04-01T17:15:04Z"}
{"meterStop":560298,"timestamp":"2025-04-01T17:15:09Z","transactionId":10038,"reason":"DeAuthorized"}
{"connectorId":1,"errorCode":"NoError","status":"Finishing","timestamp":"2025-04-01T17:15:09Z"}
```

Timple, 2025-04-01:
Update: It works! 🔌 

This was a very insight-full experience. 

I wanted to try with my real card and was looking up the `idTag`. But these are different from within my provider as seen by EVSE. (Notice the `11223344556677` turning into `DE8EFB6B`.
I always used the latter value. But with this debug logging I could see the initial value.
Setting my default card to this number actually worked 😃 

Thanks a ton!

closed 2025-04-01 by Timple