SmartEVSE/SmartEVSE-3 #81: ABB kWh meter

systeembeheerder, 2025-03-01:
**Describe the bug**
SmartEVSE reads register 0x5B0C for [currents](https://github.com/SmartEVSE/SmartEVSE-3/blob/8a60093c32c54c34d9f4018bd7d83fd3a63c1116/SmartEVSE-3/src/meter.cpp#L21C73-L21C79) for ABB kWh meters.
But [according ABB](https://library.e.abb.com/public/e4d486c4f0124484a1459431d9deae0a/Modbus%20map_Rev_B.pdf) it's an UNSIGNED int. It won't tells SmartEVSE if we are importing or exporting. Not so much an issue as EV meter but a serious one as MAINS meter.

Reading both my [ABB EV3 meters](https://search.abb.com/library/Download.aspx?DocumentID=9AKK107991A1347&LanguageCode=en&DocumentPartId=&Action=Launch) conforms this. Note that register 0x5B14 "Active Power (Watt)"  is SIGNED!

Let me know if i can help test, debug.

mstegen, 2025-03-27:
Just checked this. We already read both the current registers starting at 0x5B0C (L1-L3) and the Active power registers to determine the direction of the current in one register read operation (16 registers in total).
https://github.com/SmartEVSE/SmartEVSE-3/blob/3b6be5dbf42681b1db4b303eef8c8c654ba3109e/SmartEVSE-3/src/modbus.cpp#L352 

The actual flipping of the current sign is done here:
https://github.com/SmartEVSE/SmartEVSE-3/blob/3b6be5dbf42681b1db4b303eef8c8c654ba3109e/SmartEVSE-3/src/meter.cpp#L258

Other then that the B23 has a few extra registers, i don't see why this should not work on the EV3 meters.

mstegen, 2025-03-30:
Ok, i have received a ABB EV3 012-100 meter to play with, and to see if i can get it to work with the SmartEVSE.

Import/Export of currents works, it's correctly reported as EV and Mainsmeter.
The Energy charged is displayed wrong. The EV3 meter uses 0.0001Wh, while the B23 meter uses 0.01Wh

So far i can't recommend this meter, as there is no easy way to set the Parity from Even to None. It's only possible to set this over modbus.
Another very annoying thing is that the Address of the meter, you set on the device is entered as a decimal value from 00-99, but in reality the address is actually a hexadecimal value without any way to select A-F ??

I dont think we should officially support this meter..

closed 2025-04-01 by mstegen