SmartEVSE/SmartEVSE-3 #82: Home battery integration through sensorbox CT clamps

RobertJansen1, 2025-03-29:
**Is your feature request related to a problem? Please describe.**
Would it be possible to use the sensorbox to measure the current a home battery is charging / discharging to deduct from the p1 information?

**Describe the solution you'd like**
Add the possibility to both use p1 and ct clamps on a sensor box in order to integrate with a home battery

**Describe alternatives you've considered**
alternative would be to gather information and feed it through but that would depend on numerous systems translating / forwarding the data

**Additional context**
Add any other context or screenshots about the feature request here.

dingo35, 2025-03-29:
You are aware of the possibility to feed the battery currents via the REST and MQTT APIs?

RobertJansen1, 2025-03-29:
Yes i am aware but I would prefer "hardwiring" the information as that is more reliable and faster

RobertJansen1, 2025-03-29:
A zonneplan batterij with 3 phase ac

RobertJansen1, 2025-03-29:
so just ac ct clamps would work, and i also want the p1 metering to measure smart charging, but on top of that i want to exclude the battery power usage if that makes sense :) an alternative would be to substract a fixed 10kw (as it is mostly discharging with 10kw) when the delivery to the net is higher than 10kw