SmartEVSE/SmartEVSE-3 #94: SmartEVSE-3 charge cycle interrupted when connected to homebattery

dannk69, 2025-06-16:
Setup: SmartEVSE-3, firmware v3.7.4, with sensorbox-2 + 3 current transformers.

Since the installation a Victron homebattery (3 x Multi RS Solar + 50kwh Pylontech batteries), charge cycles are interrupted and needs to be reinitiated repeatedly. Only when charging is planned via Frank Energy “smart charging”, planned charging cycles are being completed without interruption.

Any idea, suggestion, what is causing the interruptions?

dannk69, 2025-06-17:
Yes, Frank Energy has controll over my EV wrt charging. To my knowledge only when "planned charging" is selected in the car, not when "immediate charging" is selected. In the latter case, charge cycles are interupted.  Let me double check this with Frank Energy.

Let me first update the firmware (i did last time but aparently it failed) and come back to you.

dannk69, 2025-06-19:
The interruptions apparently were caused by Frank Energy. After removing the car from the Frank Energy portal, charging cycles are no longer interrupted. Thanks for brainstorming!

Have updated the firmware to v3.7.4. Since the update, had two incidents where the SmartEVSE-3 failed to charge and was reporting “Access Denied” on the LCD-screen. Only after power cycling Sensorbox-2, the SmartEVSE-3 was ready to charge. Not sure which firmware is on the Sensorbox-2, didn’t get the SB2 WIFI message on the LCD-screen. What is triggering the "Access Denied" message?

[image]
[image]

mstegen, 2025-06-19:
Glad you solved the charge interruptions!

You can disable/enable the SmartEVSE by connecting a external button between the SW and GND terminals, and configuring the SWITCH option in the menu.
When the SmartEVSE is disabled it shows "ACCESS DENIED" on the LCD.
Pressing the "OFF" button on the website also results in the same Message.
There is a third option, that is holding the Left button on the controller. (the right button will unlock)

The Sensorbox can not interfere with this function.

mstegen, 2025-06-20:
Perfect! I'll close the issue now

closed 2025-06-20 by mstegen