SmartEVSE/SmartEVSE-3 #95: How to connect to OCCP

nekocentral, 2025-06-18:
Recently updated my SmartEVSE for to the latest version for OCCP support, But i was wondering what i need to make sure i can charge with my employers card, I know i need the NFC Reciever (Ordering that today or tomorrow), But where can i get the EVSEID i need to register to lets say Eflux, or do i need to setup an extra OCCP server for that?

dingo35, 2025-06-19:
If you search on eflux in that thread, you see that you only have to connect your SmartEVSE to their system, and then you call them so they will activate it.

nekocentral, 2025-06-19:
Wierd when i called this morning they didn't know what i was talking about will check it out a bit more

dingo35, 2025-06-19:
https://github.com/dingo35/SmartEVSE-3.5/blob/master/docs/ocpp.md

nekocentral, 2025-06-19:
Yes, got it connected using TapElectric with that, time to order the NFC reader and the energy meter.

nekocentral, 2025-06-20:
I'm having some wierd issues after enabling OCPP @dingo35 my smart keeps losing connections to my mains meter which is API fed from home assistant. What is even more wierd is that home assistant loses connection to my dsmr reader.
After disabling OCPP all connections work fine.

dingo35, 2025-06-21:
Ok thx for your report; I will look into it!

dingo35, 2025-06-22:
Ok there are some more reports on losing internet lately, but you seem to be able to reproduce it.
Can you help me by going back to older versions, and determine at which version the problem started?
Thanks in advance!!

nekocentral, 2025-06-22:
Sure, just need to set some stuff back.

nekocentral, 2025-06-22:
Current Setup:
SmartEVSE Firmware: Stable V3.7.4
SmartEVSE Intergration: 1.1.4
SmartEVSE Settings: https://gist.github.com/nekocentral/afac3ec1548f76f1ed368c94a57ecee2
HA Automation for P1 data: https://gist.github.com/nekocentral/427f7653f94a30d8e87337e5f5ec1d69

dingo35, 2025-06-22:
Why dont you interface the SmartEVSE to HA with MQTT, much lighter, much faster!

nekocentral, 2025-06-22:
Not a bad idea, i have this setup since before MQQT so never switched it
Would i just publish to current with that data?

nekocentral, 2025-06-22:
Okay got it working over MQTT, lets see if that is more stable, sorry i wasnt able to be much use for the testing. feel free to close again

closed 2025-06-22 by dingo35

nekocentral, 2025-06-22:
Just a side thing related to the modbus errors in the debug mode, just flash back to normal and ignore?
(MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
(MBhandleError)(C1) Error response: E0 - Timeout, address: 0c, function: 04, reg: 0048.