SmartEVSE/SmartEVSE-3 #97: USB UART Documentation

nekocentral, 2025-06-21:
I'm thinking of placing a Pi in my EVSE for the ISO15118 support and was wondering what other things can be set using the USB UARTport, what are things that are technically supported over UART, could i do things like have DSMR Reader on the PI to and feed it over UART to the EVSE instead of API?

mstegen, 2025-06-21:
Right now it's used as program port, and debug output.
When setting up Wifi, it's also used as backup method, to enter AP and password names.

If you want to use it for something else, remember that the USB port is not powered, and the connection you have to the ESP32 is through a CP2102 USB-Serial converter.

nekocentral, 2025-06-22:
Good to know, how are you doing the ISO15118 ESP connection then? Just using that port but with firmware that accepts the information using debug commands?

closed 2025-08-26 by dingo35