SmartEVSE/SmartEVSE-3 #99: OCPP Disconnection issues

nekocentral, 2025-07-08:
**Describe the bug**
Every day the OCPP socket disconnects, SmartEVSE says it is still connected. Looking at the backoffice my EVSE is offline. Looking at the logs in the backoffice after some time the EVSE stops sending the OCPP heartbeats. When it gets in this state it also stops authenticating against the backoffice

**Upload your config**
{"version":"v3.7.4","serialnr":6296,"mode":"OFF","mode_id":0,"car_connected":false,"wifi":{"status":"WL_CONNECTED","ssid":"Neko IOT Outside","rssi":-68,"bssid":"36:EE:52:4F:5E:3C"},"evse":{"temp":27,"temp_max":65,"connected":false,"access":false,"mode":1,"loadbl":0,"pwm":1024,"custombutton":false,"solar_stop_timer":0,"state":"Ready to Charge","state_id":0,"error":"None","error_id":0,"rfid":"Ready to read card"},"settings":{"charge_current":60,"override_current":0,"current_min":6,"current_max":15,"current_main":35,"current_max_circuit":16,"current_max_sum_mains":600,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":4,"solar_stop_time":10,"enable_C2":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0,"lcdlock":0},"mqtt":{"host":"10.4.12.13","port":1883,"topic_prefix":"SmartEVSE-45224","username":"smartevse","password_set":true,"status":"Connected"},"ocpp":{"mode":"Enabled","backend_url":"wss://ocpp.tapelectric.app/j2fvntqg4t2t","cb_id":"Neko123","auth_key":"","auto_auth":"Disabled","auto_auth_idtag":"0421A37A0A1F91","status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Eastron1P","address":12,"import_active_power":0,"total_kwh":127.5999985,"charged_kwh":4.800000191,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":127.5999985,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":40,"L1":40,"L2":0,"L3":0,"last_data_update":1751974607,"original_data":{"TOTAL":40,"L1":40,"L2":0,"L3":0}},"backlight":{"timer":0,"status":"OFF"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":255,"B":0},"smart":{"R":0,"G":255,"B":0},"solar":{"R":255,"G":170,"B":0},"custom":{"R":0,"G":0,"B":255}}}

**Upload a debug log**
N/A

**To Reproduce**
1. Connect to backoffice (Example Tap Electric)
2. Start a session and end it.
3. Wait for heartbeats to stop

**Expected behavior**
EVSE continuous to send heart beats and allows to authenticate RDID tag

**Screenshots**

[image]

[image]

dingo35, 2025-07-08:
Ok I have two tests running, one on v3.7.4 and one on v3.8.5

nekocentral, 2025-07-08:
This is my current uptime logging from the backoffice

[image]
The SmartEVSE is Line of Sight with the AP

dingo35, 2025-07-08:
I can reproduce the "loss" of heartbeat, but even when the heartbeat is lost, I can just start and stop the transaction.... 

Does your online/offline connection history mean it comes online automatically, or is there some "action" that gets it back online (e.g. start of charge, or rebooting the SmartEVSE)?

dingo35, 2025-07-13:
Can you run a continuous ping to your SmartEVSE to see if the disconnect has anything to do with wifi connectivity?

dingo35, 2025-07-13:
Another thing you can try:
pipx install esptool
-if you look in partitions_custom.csv, you see we have a partition called SPIFFS; don't be fooled, it is a LittleFS partition, but it is called SPIFFS for historical reasons; the partition starts at 0x370000 and has size 0x90000 .
-we are going to save this partition:
esptool read-flash 0x370000 0x90000 /tmp/save.bin
-now we are going to clear this partition:
esptool erase-region 0x370000 0x90000

-after restart you should notice your OCPP settings are gone; if you fill your settings you should be "starting from zero".

In case of problems you restore your save.bin:
esptool write-flash 0x370000 save.bin

If this solves your problem, please gzip and upload your save.bin so we can analyze it.

nekocentral, 2025-07-13:
> Can you run a continuous ping to your SmartEVSE to see if the disconnect has anything to do with wifi connectivity?

Will start it now, for how would you recommend?

dingo35, 2025-07-13:
Until you have an OCPP disconnect; I'd like to see your ping continuing when the OCPP is disconnecting, so we can rule out LAN/wifi problems...

nekocentral, 2025-07-13:
Okay, restarted OCPP and started an inf ping
lets see if there is any ping long

nekocentral, 2025-07-14:
Update, OCPP has been working for 20 hours since starting an inf ping to the evse.

dingo35, 2025-07-14:
Looks like you have some sort of "keepalive" problem with your network?!

dingo35, 2025-07-14:
Your RSSI is -68 so should be strong enough.
So the OCPP heartbeat keeps arriving at TapElectric too?
At what rate? Are they equidistant in time?

nekocentral, 2025-07-14:
Yea the EVSE is in line of sight with the AP, the heartbeat arrives every 15 mins and the meter value every minute

[image]

nekocentral, 2025-07-14:
Garage AP is a TP-Link EAP245(EU) v3.0 die uplinkt met mijn huis AP wat een TP-Link EAP620 HD(EU) v1.0
Signaal tussen die 2 is -86 dBm op 5ghz

dingo35, 2025-07-14:
Thats your problem; -86dBm is very close to the signal/noise ratio of current chipsets.

Try moving the inter-AP link to 2.4Ghz; you will have stronger signal and only a small loss of bandwidth, if any with such a weak signal.

nekocentral, 2025-07-14:
looks like it can only be meshed on 5ghz, might check out the SNR to my AP in the house from the evse and see if that is good eneugh

dingo35, 2025-08-26:
Closed due to inactivity.

closed 2025-08-26 by dingo35